adafruit/circuitpython #1834: Driver for MAX31850K

caternuson, 2019-04-29:
This:
https://www.adafruit.com/product/1727

Or does standard Dallas 1-wire work?

Semi-related:
https://forums.adafruit.com/viewtopic.php?f=19&t=150205

ladyada, 2019-04-29:
i think you need to make some mods to the DS18b20 library code so yeah it'd be a whole new lib

ladyada, 2019-04-29:
also note that the 1 wire circuitpython libs dont work on linux at all because linux does 1wire at the kernel level

Billdapart, 2022-12-28:
It's been quite a while since this was brought up so sorry to dredge it up but are there any plans to build a library for this sensor and circuit python? If not, I am going to give it a go but it is definitely going to be an uphill battle for me so didn't want to duplicate efforts if it was already planned/underway. Thanks for all the awesome hardware!

ladyada, 2022-12-30:
nobody has taken it on that we know of - go for it!

caternuson, 2023-01-02:
The DS18B20 library may help as a general reference and onewire example:
https://github.com/adafruit/Adafruit_CircuitPython_DS18X20/blob/main/adafruit_ds18x20.py