adafruit/circuitpython #1835: CircuitPython Workshops in Israel

urish, 2019-04-29:
I'm running a popular meetups group in Israel called [IoT Makers](https://www.meetup.com/IoT-Makers-Israel/). We organize hands-on workshops for the Maker community, and would love to introduce CircuitPython to our workshops. In order to make this possible, we'd need your help:

1. By adding support for the Thingy:52 and its peripherals. We have 24 Thingy:52 devices that were generously contributed to our community by Nordic Semiconductor.
2. Alternatively, if Adafruit could sponsor some Circuit Playground boards for this purpose, that would be a great help and an enabler for this workshops.

I'd love to hear your thoughts and collaborate this. Let's make this a reality 👍

hexthat, 2019-05-01:
I don't think nRF52832 is supported currently, but if you look at 3.x it was once.

ptorrone, 2019-05-01:
@urish please email pt@adafruit.com and we'll send ya out some boards.

urish, 2019-05-01:
@ptorrone Done! Thx 👍

tannewt, 2019-05-02:
Looks like this has been resolved. :-)

closed 2019-05-02 by tannewt