adafruit/circuitpython #1839: displayio.Group scale drawing x=y=0 edge pixels twice

caternuson, 2019-05-02:
Test code below. Draws a bitmap with a 1 pixel border then scales it up. Scaled result border is not symmetric. The top (y=0) and left (x=0) edges appear doubled?

```python
import board
import displayio

SCALE = 8
WIDTH = 10
HEIGHT = 10
X = 20
Y = 20

bitmap = displayio.Bitmap(WIDTH, HEIGHT, 2)

palette = displayio.Palette(2)
palette[0] = 0xFFFFFF
palette[1] = 0xFF0000

for x in range(WIDTH):
    for y in range(HEIGHT):
        if (x == 0 or x == WIDTH - 1 or
            y == 0 or y == HEIGHT -1 ):
            bitmap[x, y] = 0
        else:
            bitmap[x, y] = 1

tilegrid = displayio.TileGrid(bitmap, pixel_shader=palette)

group = displayio.Group(scale=SCALE, x=X, y=Y)
group.append(tilegrid)

board.DISPLAY.show(group)
```

Result:
[image]

charlesbmale, 2019-05-06:
I do not know which version of Circuit Python you were using but for CircuitPython 4.0.0-rc.1 I received a similar issue with the Test Code you provided and I took that code and fixed the double edged borders with the following code. Hope it helps!!

import board
import displayio

SCALE = 8
WIDTH = 10
HEIGHT = 10
X = 20
Y = 20

bitmap = displayio.Bitmap(WIDTH, HEIGHT, 2)

palette = displayio.Palette(2)
palette[0] = 0xFFFFFF
palette[1] = 0xFF0000

for x in range(WIDTH):
    for y in range(HEIGHT):
        if (x == 0 or x == WIDTH - 1 or
            y == 0 or y == HEIGHT -1 ):
            bitmap[x, y] = 0
        else:
            bitmap[x, y] = 1

tilegrid = displayio.TileGrid(bitmap, pixel_shader=palette, x=X, y=Y)

group = displayio.Group(scale=SCALE)
group.append(tilegrid)

board.DISPLAY.show(group)

while True:
    pass

caternuson, 2019-05-06:
Oops, I should have specified CP version. Thanks for mentioning that. I was using 4.0 RC 1.

I tried above and it does seem to render correctly. It looks like all you've done is move the positional information from `group` to `tilegrid`. Was trying to set position via the `group` an incorrect usage?

charlesbmale, 2019-05-06:
When tackling your issue I copy and pasted your code and tried to replicate the issue you were having and it displayed a position argument error on line 24. I fixed that error but it turned out that I was using an older version. So when I updated the version, the code I had didn't work but still received a position error. So I copy and pasted the code you had again and received the same issue you had. I assumed the reason your code didn't work was due to the way you set the position (since I received a position error earlier) so I fixed the position again and it worked. Sorry for the long response, I just wanted to make sure I specified as much as possible

caternuson, 2019-05-06:
There have been some changes to the positional argument as 4.0 has moved through the beta releases. At one point it was a tuple. But now both `Group` and `TileGrid` have specific and separate `x` and `y`. 
https://circuitpython.readthedocs.io/en/latest/shared-bindings/displayio/TileGrid.html
https://circuitpython.readthedocs.io/en/latest/shared-bindings/displayio/Group.html
So I think either (or both) can generally be used to affect position. The fact that the issue only shows up when setting position via `Group` might be indicating something.

makermelissa, 2019-05-06:
Yep, I helped him with that change.

PR #1872 (2019-05-10): Handle -scale to 0 correctly in Group
Fixes #1839
commit e64bbc41 (2019-05-10): Handle -scale to 0 correctly in Group

Fixes #1839
closed 2019-05-10 by dhalbert