adafruit/circuitpython #1841: RuntimeError about SDA/SCL pullup

jdemilledt, 2019-05-05:
Upon initializing the I2C bus with my Trinket M0, I get an error complaining that SDA or SCL is not pulled up.

This seems odd, however, since I have resistors on both lines and I've tested them with a multimeter to be sure. How could I troubleshoot this?

hulkco, 2019-05-05:
You can see the guide on how to do it here: [https://learn.adafruit.com/circuitpython-essentials/circuitpython-i2c](https://learn.adafruit.com/circuitpython-essentials/circuitpython-i2c)
[image]

ladyada, 2019-05-05:
hiya it could be a wiring issue, you can post to the forums, but the detection works pretty well so its unlikely to be a core bug. please post to the adafruit forums and/or check your wiring! 
closing for now

closed 2019-05-05 by ladyada

jdemilledt, 2019-05-05:
I do have the pullups (4.7K), and I made sure to test them with a multimeter, including continuity, so I know the wiring is working.

caternuson, 2019-05-05:
Please post photos of your setup in the forums:
https://forums.adafruit.com
We can not recreate this issue.

[image]
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Trinket M0 with samd21e18
>>> import board, busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> 
```

remove resistors:
```python
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Trinket M0 with samd21e18
>>> import board, busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: SDA or SCL needs a pull up
>>> 
```

jdemilledt, 2019-05-05:
I'm running on 4.0rc1 FYI.

caternuson, 2019-05-05:
4.0 RC 1 checks.

```python
Adafruit CircuitPython 4.0.0-rc.1 on 2019-04-23; Adafruit Trinket M0 with samd21e18
>>> import board, busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>>
```
remove resistors:
```
Adafruit CircuitPython 4.0.0-rc.1 on 2019-04-23; Adafruit Trinket M0 with samd21e18
>>> import board, busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: SDA or SCL needs a pull up
>>> 
```

deshipu, 2019-05-05:
@jdemilledt perhaps you could post a photo of your setup?

jdemilledt, 2019-05-05:
https://forums.adafruit.com/viewtopic.php?f=60&t=151338 I've made a forum post.

Spongman, 2023-12-16:
why is this an error on rp2040 boards? all those pins have internal pull-ups. surely this error is unnecessary. if the pin isn't high, just enable the pull-up.

deshipu, 2023-12-16:
The internal pull-ups are way too weak for I2C.