adafruit/circuitpython #1843: An Idea on Bluetooth to MQTT

Tasm-Devil, 2019-05-05:
Hi there.
I have a question regarding mqtt and usocket.
The old Micropython mqtt-library uses the class usocket.

Now there is no MQTT library in the nrf52840 and no usocket library, because the chip has no Ethernet.

What if you use Nordic UART to build an MQTT connection with a broker? Would not that be great.

It is not theoretically possible to write a usocket class that can connect to a server (MQTT broker) via the UART interface.
A umqtt class (simple.py, robust.py) could then use this socket.

Of course, the third thing you need is a software that builds a proper socket connection to the Internet. But this could work on any computer with Blutooth 4. You could rewrite WebbleMQ so that it does not use mosquitto.js but only builds a websocket connection to an MQTT-Brocker.

What do you all mean. Would it be possible?

ladyada, 2019-05-05:
sure, you could try porting umqtt to the airlift board, it provides some `socket` ability

Tasm-Devil, 2019-05-05:
I think, it was probably a bad idea to use the socket module.

I tried creating an fake-socket-module to test [simple.py](https://github.com/micropython/micropython-lib/blob/master/umqtt.simple/umqtt/simple.py) but now i think, its better to do the mqtt-protocol stuff in the javascript file and not on the board. This makes it possible to use the [wss](https://en.wikipedia.org/wiki/WS-Security) protocol.

But its still a lot work to be done and i am no javascript expert. If there is anyone out there. Help would be great. Starting point could be [WebbelMQ](https://github.com/Tasm-Devil/webbleMQ)

ladyada, 2019-05-05:
sounds like a non-circuitpython project/issue then? you want someone to develop a webservice?

Tasm-Devil, 2019-05-05:
Yes and No :)
On the Circuitpython side, we still need a mqtt-module, that uses the ble UARTServer.
I'm working on it

ladyada, 2019-05-05:
ok sounds good! we have a library guide here https://learn.adafruit.com/creating-and-sharing-a-circuitpython-library that can help you get it into good shape!

Tasm-Devil, 2019-05-07:
ok thanks for the link. I will consider doing it the right way when this grows up.
Here is my first demo of a [BLE2MQTT](https://github.com/Tasm-Devil/BLE2MQTT) relay service for Circuitpython.

closed 2019-05-07 by Tasm-Devil