adafruit/circuitpython #1857: Float function rounds to two decimal places.

dherrada, 2019-05-07:
I've tried this in 3.1.1 on multiple boards and 4.0.0-rc1 on the feather M4. What happens is:
```
>>> float('8051.4091')
8051.41
```
And the same thing happens when I run that function from a file

dherrada, 2019-05-07:
This is probably a hardware limitation since how accurate float() is depends on the chip itself. One possible solution could be to add a different float function with 2 arguments to allow people to specify how long they want their float to be.

barbudor, 2019-05-07:
@dherrada this is nothing to do with `float()` doing some rounding to N decimal place. This is intrinsic to single precision float.
It is not possible to represent 8051.4091 in, single precision float due to mantissa limited to 23 bits.

Note that (Circuit)Python already support a `round( value [,ndigits])`built-in for rounding.

dherrada, 2019-05-08:
Ok. Thanks for the clarification. I was trying to implement what you suggested for rounding but the problem with using round() was that round() can't round a string without having to convert it to a float with float() first.

closed 2019-05-08 by tannewt