adafruit/circuitpython #1864: pybadge: games reset the display after 2-10 seconds

deshipu, 2019-05-08:
As we have the ``_stage`` module to work with displayio, and the ``stage`` and ``ugame`` modules frozen on the pybadge, all of the µGame games should JustWork™ on it. And that is indeed how it is on the version D prototype of pybadge that I have and that I have tested it all with. It also works with itsybitsy m4 express with a display connected, and with the minitft featherwing on a feather m4 express, and even on hallowing and of course the µgame.

However, when I asked people to try those games on the final versions of pybadge, they would get a flash of the game's screen, followed by a white screen. We tried different versions of the ``ugame`` module, performing a full reset and initialization of the display (https://github.com/python-ugame/circuitpython-stage/commit/7744365d887df33a04e25fdec72cc610a5cb0d08), manual reset with the reset pin (https://github.com/python-ugame/circuitpython-stage/commit/bae3a0952f28e7e8ba36400eb724e76068d91456), etc. — but only after we decresed the SPI clock to 12, we noticed the screen now goes white not immediately, but after 2-10 seconds of working properly.

As far as I know, the only difference with the version I have is that the final version has ``PA00`` and ``PA01`` swapped — those are ``TFT_LITE`` and ``TFT_RST`` pins. It looks to me like the reset pin is getting pulled low at some point, resetting the display without reinitialization, resulting in it going white. I suspect this might be due to some kind of cross-talk between the data and rst pins, as it becomes less likely at lower communication speed, and doesn't happen at all when there are no constant animations, but without a board to test I can't really be sure. It could also be a result of the old RST pin still being hardcoded somewhere and getting pulled low for some reason, or even another memory corruption bug.

deshipu, 2019-05-08:
The problem can be reproduced by flashing pybadge with version 4.0.0-rc-1 of CircuitPython firmware, copying the ``jumper.py``, ``level.bmp``, ``walls.bmp`` and ``jumper.bmp`` files onto its disk, and renaming the ``jumper.py`` to ``main.py``.

tannewt, 2019-05-09:
So the version that does a re-init mostly works. I think the checked in version suffers from leaving reset floating and therefore very susceptible to cross talk. It doesn't look like a hardware issue to me. (And therefore you should be able to reproduce with yours as well.)

It looks like it's still fighting the native screen update where the terminal is still attempting to show on the screen. `_stage` should set the display to showing it own empty group so that prints don't trigger a display update.

Sorry but I can't spend more time on this. I can't support two different gaming APIs. I'd rather get the native groups rendering fast. So, I'll leave this open but under long-term.

deshipu, 2019-05-09:
@makermelissa reproduced the problem with all versions, no matter what state the reset pin was left in.

I'm closing this issue, but feel free to re-open it if the official api gets fast enough to have the same problem.

closed 2019-05-09 by deshipu