adafruit/circuitpython #1871: Sort filenames in .po translation files

dhalbert, 2019-05-09:
`xgettext --add-location=file` does not sort the filenames: they seem to be in random order, which causes unnecessary diffs. We could post-process the files to sort the filenames.

I looked in the `gettext` [bug list](https://savannah.gnu.org/projects/gettext/) and don't see this mentioned in any open or closed bug.

godlygeek, 2019-05-13:
Any chance this is just because we just pipe the output of `find` directly to `xargs xgettext` and `find` returns files in an arbitrary order?  Perhaps piping through `sort` in the middle would do it.

tannewt, 2019-05-13:
It's worth a shot!

dhalbert, 2019-05-13:
I was going to write an awk script or similar, but then discovered that the filenames could be split over multiple lines.

tannewt, 2019-05-14:
I'm doing this. I think @godlygeek is right.

smurfix, 2019-05-14:
Either that, or freeze PYTHONHASHSEED.

tannewt, 2019-05-14:
@smurfix How does PYTHONHASHSEED fit into it?

commit 6756821f (2019-05-14): Update translations and sort input files

This should stabilize the sort of files in the .po files.

Fixes #1871
closed 2019-05-14 by dhalbert