adafruit/circuitpython #1875: 4.0.0 RC2 - Side-effects of https://github.com/adafruit/circuitpython/pull/1860 don't allow assigning various cross-types

ladyada, 2019-05-11:
some assignments such as bool to bytearray no longer work, but do in cpython3

```$ python
Python 3.7.2 (default, Dec 26 2018, 08:10:46)
[GCC 7.4.0] on msys
Type "help", "copyright", "credits" or "license" for more information.
>>> a = bytearray(3)
>>> print(a)
bytearray(b'\x00\x00\x00')
>>> a[0] = True
>>> print(a)
bytearray(b'\x01\x00\x00')
>>>
```
vs

```
Adafruit CircuitPython 4.0.0-rc.2 on 2019-05-10; Adafruit Feather M4 Express wit
h samd51j19                                                                     
>>> a = bytearray(3)                                                            
>>> print(a)                                                                    
bytearray(b'\x00\x00\x00')                                                      
>>> a[0] = True                                                                 
Traceback (most recent call last):                                              
  File "<stdin>", line 1, in <module>                                           
OverflowError: value would overflow a 1 byte buffer  
```

dhalbert, 2019-05-11:
@tannewt I'm thinking we should revert this PR for now and revisit it for 4.1, because it's breaking known and unknown things, and it's not a vital feature for 4.0.0 (it should have been added in a beta).

dhalbert, 2019-05-11:
I'm working on fixing this instead of reverting. I'm seeing if I can simplify it too.

closed 2019-05-13 by tannewt

kevinjwalters, 2019-06-22:
There's a sine wave function somewhere that's blowing up due to 65536 going into an `"H"` `array.array`. Is this in the same vein as what @dhalbert was commenting on.

Discussed in: https://forums.adafruit.com/viewtopic.php?f=60&t=153174

PR #1879 (2019-05-13): Handle truth values; speed up smallint checks
Fixes #1875.

- Converts truth values to 0 and 1 appropriately. #1860 introduced a regression that made them not valid.
- Separates out checking small int ranges to speed up the common cases.

@godlygeek You may want to take a look.