adafruit/circuitpython #1886: Add check to travis to make sure new boards are built

tannewt, 2019-05-14:
Alternatively, have Travis always build everything.

C47D, 2019-07-12:
I assume we will need to keep track of the built assets versus the boards list?

tannewt, 2019-07-15:
Right, it should cause an error if a board directory is added but not built.

C47D, 2019-07-15:
Did a quick check of the `travis.yml` and I think the check needs to be added on the `tools/build_release_files.py` script, is it?

I just want to make sure I'm on the right track.

tannewt, 2019-07-15:
I'm not sure where it would be done.

sommersoft, 2019-07-17:
After some mental walking through the problem, I don't think `tools/build_release_files.py` is a good home for this.

Since the Travis jobs run as a matrix, and `build_release_files` only gets a limited set of board names, it would be difficult (but not impossible) to keep track of what matches the condition of `board not built, but exists`. (Further explanation if you don't understand the `.travis.yml`. Each line under `env:` is an environment variable, and forces a separate matrix job. Those env vars are accessed by `build_release_files` with `os.environ["TRAVIS_BOARDS"]`. So, each time the script is run in each matrix job, it doesn't have the "full" list.)

I would suggest a new script. It could be Python, or even Bash. If Python, it could still use the `build_boards_info.get_board_mapping()` as a shortcut to get the list of boards in `ports/`. Then you could just have the script open `.travis.yml` and extract all of the boards listed in the `TRAVIS_BOARDS` env vars. I've been on a Regex kick lately; here's a search pattern that works: `TRAVIS\_BOARDS\=\"(.+)\"`. Then its just checking that everything returned from `get_board_mapping()` is in the Regex results.

Lastly, I would run this new script in the `before_script:` section in `.travis.yml`. It only needs to run once, and there's no sense in building everything else if this will fail anyway.

C47D, 2019-07-18:
Hi @sommersoft, thanks for the explanation, now the problem to solve and the solution you're proposing are clear to me, i just have one question. Does the result of the search pattern `TRAVIS\_BOARDS\=\"(.+)\" the same as if i get the boards using `os.environ["TRAVIS_BOARDS"]`?
I'm not very familiar with regex, but if it's necessary i can take a look at it.

Meanwhile, what do you think it should be a good name for the new script? I think `check_all_boards_are_being_built.py` is a long one, any suggestions?

C47D, 2019-07-19:
Hi,

Had to delete my old comment because i didn't read properly your reply, so all my questions are now answered.

commit 8eab946c (2019-07-29): Merge pull request #2003 from C47D/travis_check_new_boards

Add check to travis to make sure new boards are built, fix #1886
commit a1db6c43 (2019-07-29): [Draft] Add check to travis to make sure new boards are built, fix #1886
closed 2019-07-29 by sommersoft