adafruit/circuitpython #1890: round(<integer>, <n>) throws NotImplementedError

dhalbert, 2019-05-16:
Reported by forum user lawrenceyy in https://forums.adafruit.com/viewtopic.php?f=60&t=151793:

`round(<integer>, <n>)` throws `NotImplementedError`. User found this via `round(0, 2)`. `round(<float>, 2)` is OK, as is `round(<integer>)`.

This was working in 3.1.2, but broke as early as 4.0.0-beta.2. MicroPython 1.9.4 seems OK.

dhalbert, 2019-05-16:
This has to do with the default setting of `MICROPY_PY_BUILTINS_ROUND_INT`, which was added by https://github.com/micropython/micropython/commit/b318ebf1015ced6354f8bbaf035308214b3f5c5d. The original idea was to add handling negative rounding of integers, like `round(123,-1)`, but it took significant extra code, so the flag was added. I'm not sure it was Damien's intention to disallow two-arg rounding of integers completely. I filed an issue there and will see what he says. In the meantime, we might turn this on, or just fix it ourselves if it grows the code too much.

dhalbert, 2019-05-16:
@tannewt Enabling `MICROPY_PY_BUILTINS_ROUND_INT` adds 151 bytes to image. Fixing the code so that `round(x, <negative n>)` is still not implemented but `round(<int>, <positive n>) works adds 16 bytes.

Current trinket build with French translation has 1264 bytes free with this turned on. That may be the largest one right now.

tannewt, 2019-05-17:
I'm fine adding it back.

dhalbert, 2019-08-01:
Fixed by #1896.

PR #1896 (2019-05-21): Enable MICROPY_PY_BUILTINS_ROUND_INT; make round() work better when flag is disabled
Fixes #1890.

Set `MICROPY_PY_BUILTINS_ROUND_INT` = `1` for all builds, which implements `round(n, <negative ndigits>)`, so that, for instance `round(189, -1)` will round to the nearest 10.

Also tweaks the implementation so that even if `MICROPY_PY_BUILTINS_ROUND_INT` = `0`, `round(<int>, <positive ndigits>)` will still work. Previously rounding an int with a specified ndigits would throw `NotImplementedError`.

Adds 151 bytes or so to a build.
closed 2019-08-01 by dhalbert