adafruit/circuitpython #1891: Some boards unintentionally sharing USB_PID values

kevinjwalters, 2019-05-16:
I was reviewing USB PID allocations, as you do, and noted that the PyPortal and GrandCentral M4 use the _same_ one, **0x8032**. @tannewt confirmed in discord that this sharing is incorrect.

Here's a summary of all of them in PID order generated from the `ports/atmel-samd/board` directory:

```
$ date ; grep USB_PID $(grep 'USB_VID.*=.*0x239[aA]' -l */*.mk) | awk -F: '{ print $2, $1; }' | sort -k 3
Thu 16 May 10:58:36 BST 2019
USB_PID = 0x8012 itsybitsy_m0_express/mpconfigboard.mk
USB_PID = 0x8014 metro_m0_express/mpconfigboard.mk
USB_PID = 0x8015 feather_m0_adalogger/mpconfigboard.mk
USB_PID = 0x8015 feather_m0_basic/mpconfigboard.mk
USB_PID = 0x8015 feather_m0_rfm69/mpconfigboard.mk
USB_PID = 0x8015 feather_m0_rfm9x/mpconfigboard.mk
USB_PID = 0x8019 circuitplayground_express_crickit/mpconfigboard.mk
USB_PID = 0x8019 circuitplayground_express/mpconfigboard.mk
USB_PID = 0x801D gemma_m0/mpconfigboard.mk
USB_PID = 0x801D pewpew10/mpconfigboard.mk
USB_PID = 0x801F trinket_m0_haxpress/mpconfigboard.mk
USB_PID = 0x801F trinket_m0/mpconfigboard.mk
USB_PID = 0x801F ugame10/mpconfigboard.mk
USB_PID = 0x8021 cp32-m4/mpconfigboard.mk
USB_PID = 0x8021 metro_m4_express/mpconfigboard.mk
USB_PID = 0x8023 feather_m0_express_crickit/mpconfigboard.mk
USB_PID = 0x8023 feather_m0_express/mpconfigboard.mk
USB_PID = 0x8023 feather_m0_supersized/mpconfigboard.mk
USB_PID = 0x8023 feather_radiofruit_zigbee/mpconfigboard.mk
USB_PID = 0x8026 feather_m4_express/mpconfigboard.mk
USB_PID = 0x8028 pirkey_m0/mpconfigboard.mk
USB_PID = 0x802C itsybitsy_m4_express/mpconfigboard.mk
USB_PID = 0x802C pybadge/mpconfigboard.mk
USB_PID = 0x802E pygamer/mpconfigboard.mk
USB_PID = 0x8030 trellis_m4_express/mpconfigboard.mk
USB_PID = 0x8032 grandcentral_m4_express/mpconfigboard.mk
USB_PID = 0x8032 pyportal/mpconfigboard.mk
USB_PID = 0x8035 arduino_mkrzero/mpconfigboard.mk
USB_PID = 0x8038 metro_m4_airlift_lite/mpconfigboard.mk
USB_PID = 0xD1ED hallowing_m0_express/mpconfigboard.mk
```

Perhaps the `itsybitsy_m4_express` sharing with `pybadge` is incorrect too? I don't know what a `pewpew10` is, is it a variant of a Gemma M0, they are sharing too?

`hallowing_m0_express` is an anomaly with its number clearly from a different range.

deshipu, 2019-05-16:
If this is a problem, I can get my own USB VID&PID pair for ugame10 and pewpew10 — however, I'm not sure if the CircuitPython drivers for Windows will then work for them?

dhalbert, 2019-05-16:
@deshipu I don't think that's really an issue. No host program I know of uses the PID for significant things except maybe Arduino.

dhalbert, 2019-05-16:
I'll work up a PR for this. @kevinjwalters The Hallowing PID is a joke :)

deshipu, 2019-05-16:
@dhalbert Mu uses VID/PID to find the devices.

deshipu, 2019-05-16:
I will apply for the new PIDs then.

dhalbert, 2019-05-16:
@deshipu Mu now uses a wildcard: All boards with the Adafruit VID (0x239a) are recognized.

kevinjwalters, 2019-05-17:
In Windows I see my new PyPortal boards (running 4.0.0 rc3) as `@oem31.inf,%grand_central_m4_express% circuitpython (8032:00);Adafruit Grand Central M4 Express CircuitPython (8032:00)`. Presumably that's due to the clash?

dhalbert, 2019-05-17:
No, that's a weirdness of driver installation on certain versions of Windows. What version of Windows are you using?

kevinjwalters, 2019-05-17:
Windows 8.1 with the Adafruit 2.3.4.0 drivers.

dhalbert, 2019-05-17:
Yes, that's some mild driver name text formatting bug in Windows 8 and up given certain .inf files. It's not substituting a definition in the .inf file properly. It works in Windows 7.

dhalbert, 2019-05-17:
@kevinjwalters OK, sorry, misread your error report. Yes, the reporting of it as a Grand Central is due to PID mismatch. However, all the CircuitPython boards install the same driver, so it doesn't really matter. I'll need to update the drivers at some point to add new boards.

If you are at all able to upgrade to Windows 10, I would recommend it, since all this would go away. However, I can understand if your machine is stuck on that or it's personal preference or admin reasons.

Does MIDI work properly on 8.1 on CPX, etc.?

kevinjwalters, 2019-05-17:
Yep, MIDI is fine. The only issue I have is with the boards (plural) all calling themselves `CircuitPython MIDI`, discussed in https://forums.adafruit.com/viewtopic.php?f=60&t=151271. Some things can cope with that to some degree ([Ableton Live](https://forum.ableton.com/viewtopic.php?f=1&t=234679)), other less sophisticated products can't. It would be nice if that could be specified by the programmer / user.

dhalbert, 2019-05-17:
Fixed by #1892.

PR #1892 (2019-05-16): correct USB PIDs for several boards
Fixes #1891. Checked against master internal list. @ladyada tagging you for info.
closed 2019-05-17 by dhalbert