adafruit/circuitpython #1905: Cannot paste unicode into the REPL

tannewt, 2019-05-22:
It should echo back unicode characters just fine even if they don't show on the display's terminal.

DavePutz, 2020-02-16:
For UTF-8 characters the fix seems simple: In lib/mp-readline/readline.c replace line 209
   } else if (32 <= c  && c <= 126) {
 with
   } else if (32 <= c ) {

This allows REPL to handle at least the UTF-8 characters
>>> s='27° C'
>>> print(s)
27° C

But there seem to be some other issues:
>>> s='27° C'
>>> s
'27 C'       // no degree symbol
>>> print (len(s))
4              //should be 5 if the degree symbol is included

And, of course, UTF-16 characters aren't covered.

Is the simple patch worth a PR, or do the other issues need to be addressed as well?

Note that CPython (tested on 3.6.8) throws an error in REPL:
>>> s='27° C'
  File "<stdin>", line 0

    ^
SyntaxError: 'utf-8' codec can't decode byte 0xb0 in position 5: invalid start byte

tannewt, 2020-02-18:
Thanks for taking a look at this @DavePutz! I think a small simple patch that improves the current situation would be great. For the other issues we can file separate issues and/or PRs.

I wouldn't worry about UTF-16. UTF-8 is most common.

dhalbert, 2020-02-19:
Fixed by #2631.

closed 2020-02-19 by dhalbert