adafruit/circuitpython #1912: PyPortal RTC/time stops advancing when I2C device connected

bbtinkerer, 2019-05-24:
On the PyPortal if I connect any powered I2C device to side 4 pin connector, the RTC/time stops advancing and just pauses. The time continues again (does not restart, just picks up where it paused) after disconnecting the I2C device's power. The code I used to just check the time behavior as follows, then just power on/off the I2C device:

```python
import time

i = 0
while True:
    i = i + 1
    print("time: {0}, counter: {1}".format(time.time(), i))
```

I know the devices are correctly connected because I am able to control/read their value. I tried externally powering the I2C devices also, but still had the same behavior.

bbtinkerer, 2019-05-25:
I do not think the problem is a software problem. I changed the PyPortal voltage output on the connectors to the 3.3V by cutting the trace 5V trace on SJ1 and solder bridging to the 3.3V. The RTC works now when I2C devices are connected.

dhalbert, 2019-06-21:
This is still worth looking at, because the I2C implementation should not hang up the RTC if there's something wrong. https://forums.adafruit.com/viewtopic.php?f=60&t=151839

jerryneedell, 2019-06-21:
FWIW - I have reproduced this with a I2C soil sensor on a pyportal  -- time.time() stop incrementing. time.monotonic() continues incrementing.
Just connecting the I2C device stops time.tim() from incrementing - it does not have to be accessed.

dhalbert, 2019-06-21:
@jerryneedell Which sensor? And are you applying 5v or 3.3v via the connector?

jerryneedell, 2019-06-21:
https://www.adafruit.com/product/4026
with he jumper in the default 5V position
Also not that the sensor does not work properly. the temperature values are wrong. I don't want to repeat this test!
at 3V it works fine as reported

jerryneedell, 2019-06-21:
I hav e the jumper set to 3V for portals I use with I2C -- I have one at 5V still so I used it to check.

jerryneedell, 2019-06-21:
I may have damaged the sensor

jerryneedell, 2019-06-21:
hmm -- looking at the schematic for the pyportal -- it looks like with the jumper at 5V, it puts 5V on the I2C VCC -- SDA/SCL are pulled up to 3.3 V on the pyporta for its adt7410 temperature sensor, but the I2C device (the soil sensor) will pull them to 5V -- not sure what this does.  In any case, my soil sensor temperature values are not working properly any more.  I think I'll stop testing this.

jerryneedell, 2019-06-21:
the soil sensor that I was giving me bad readings after using 5V I2C on the pyportal still will not work with Circuitpython, but it does work under Arduino!!  I'm confused, but relieved.

jerryneedell, 2019-06-22:
After convincing myself that the soil sensor was not damaged by the 5V I2C I have resumed testing this.

I am now finding that the issue is intermittent.
Here is an example where time.time() does increment wit the I2C sensor connected

I am unable to reproduce the issue now but I have reproduced  it on other occasions. 
```

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 4.1.0-beta.0-31-g3863a6ab8 on 2019-06-20; Adafruit PyPortal with samd51j20
>>> 
>>> import timetest
temp: 16383.0  moisture: 322
time: 946684816, counter: 1
temp: 16383.0  moisture: 331
time: 946684816, counter: 2
temp: 16383.0  moisture: 332
time: 946684817, counter: 3
temp: 16383.0  moisture: 354
time: 946684818, counter: 4
temp: 16383.0  moisture: 331
time: 946684819, counter: 5
temp: 16383.0  moisture: 334
time: 946684820, counter: 6
temp: 16383.0  moisture: 337
time: 946684821, counter: 7
temp: 16383.0  moisture: 331
time: 946684821, counter: 8
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "timetest.py", line 24, in <module>
KeyboardInterrupt: 
>>> 

```

this is the code I am running
```
 
import time

from board import SCL, SDA
import busio

from adafruit_seesaw.seesaw import Seesaw

i2c_bus = busio.I2C(SCL, SDA)

ss = Seesaw(i2c_bus, addr=0x36)
i = 0

while True:
    # read moisture level through capacitive touch pad
    touch = ss.moisture_read()

    # read temperature from the temperature sensor
    temp = ss.get_temp()

    print("temp: " + str(temp) + "  moisture: " + str(touch))
    i = i + 1
    print("time: {0}, counter: {1}".format(time.time(), i))
    time.sleep(1)

```

jerryneedell, 2019-06-22:
I created a dummy I2C device with just 10K resistors pulled to VCC. Runnig this code
```
 
import time
i = 0
while True:
    i = i + 1
    print("time: {0}, counter: {1}".format(time.time(), i))
    time.sleep(1)
```

connecting the "device" to the Pyportal with the 5V I2C you can see that the rime increment stops
```

time: 946685538, counter: 391
time: 946685539, counter: 392
time: 946685540, counter: 393
time: 946685541, counter: 394

time: 946685542, counter: 395
time: 946685542, counter: 396
 connected "device"
time: 946685542, counter: 397
time: 946685542, counter: 398
time: 946685542, counter: 399
time: 946685542, counter: 400
time: 946685542, counter: 401
disconnected  "device"

time: 946685543, counter: 402
time: 946685544, counter: 403

```

This does not happen on a PyPortal with the 3,3V jumper in place for I2C

jerryneedell, 2019-06-22:
I was not able to reduce this on a feather_m4_express.
I tried both with just the Dummy device with its VCC connected to VUSB
and with th dummy device and an ADT7410 connected (like it is on the Py Portal.
The time.time() kept incrementing and the adt7410 works normally.

jerryneedell, 2019-06-22:
hmmm -- if I connect my "dummy" device to the PyPortal -- it takes it along time to boot -- much longer than normal
 ~10 sec or more

jerryneedell, 2019-06-22:
and another observation:
releasing either SDA or SCL starts the time.time() incrementing again. It only stops if both are pulled up.

ladyada, 2019-06-22:
make sure the i2c pins are pulled up to 3V not 5V

jerryneedell, 2019-06-22:
Right -- this only happens if you connect an I2C device to a "default" PyPortal where I2C VCC is set to 5V on the JST connector

ladyada, 2019-06-22:
yeah thats a known issue with the samd51 - it does not like pullups to 5V

jerryneedell, 2019-06-22:
A more prominent warning may be needed for PyPortal -- this comes up a lot.

ladyada, 2019-06-22:
sure, where do you think it should go?

jerryneedell, 2019-06-22:
I guess here https://learn.adafruit.com/adafruit-pyportal/pinouts#i2c-connector-2-18  it says you can set the jumper -- may want say you "should" set the jumper for I2C devices

dhalbert, 2019-06-22:
@ladyada Maybe next board rev should change the jumper to default 3.3V.

ladyada, 2019-06-22:
the next board rev will have a level shifter (its in the mini version already)

ladyada, 2019-06-22:
added!

jerryneedell, 2019-06-22:
Should we close this? Just say "don't do that"  ;-)

dhalbert, 2019-06-22:
Yes, it's "solved". "If it hurts, don't do that." Since the guide has been updated, this is enough. But thanks for testing.

closed 2019-06-22 by dhalbert

ladyada, 2019-06-22:
yep future boards don't have this issue - e.g. the pygamer/pybadge have proper level shifting. things get learned! :D