adafruit/circuitpython #1913: Crash into the HardFault_Handler

iraytrace, 2019-05-24:
Got a hard crash
[bug.zip](https://github.com/adafruit/circuitpython/files/3217451/bug.zip)

dhalbert, 2019-05-24:
The BOOT_OUT.TXT file in the zip file says it's beta 7. Could you upgrade to 4.0.1 and try again?

tannewt, 2019-06-13:
@iraytrace any update on this?

DavePutz, 2019-08-08:
Running some diagnostics this looks like a crash due to memory fragmentation. I believe this is due
to having 'import' statements scattered throughout the code. If all imports are placed at the top of the Python script things seem to run OK. i.e.:
```
import board
import digitalio
import time
from adafruit_debouncer import *
import neopixel
import os
from DiscJockey import *
import pulseio
```
In addition to helping memory management, this is just good Python programming practice. As Python PEP08 states:

> Imports are always put at the top of the file, just after any module comments and docstrings, and before module globals and constants.

kevinjwalters, 2019-08-08:
I'm not disagreeing with (non conditional) `import`s at the top but shouldn't running out of memory give a `MemoryError` exception?

DavePutz, 2019-08-08:
Sorry - I should have made my comment clearer. MemoryError exceptions are what I was seeing when I was doing the requested testing using the 4.0.1 release. I did not test with the beta version the issue was reported on. The error occurs at line 96, which is the DiskJockey import statement.

dhalbert, 2020-11-01:
Closing due to age of versions that showed an error.

closed 2020-11-01 by dhalbert