adafruit/circuitpython #1917: Crash at HardFault_Handler

jnalezny, 2019-05-28:
I was playing around with a circuit playground express and a crickit.  I got this error

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard.  Whoops!
Please file an issue with.....
with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler

My code:
`import time
from adafruit_crickit import crickit
#from adafruit_motor import stepper

print("Bi-Polar or Uni-Polar Stepper motor demo!")

# make stepper motor a variable to make code shorter to type!
stepper_motor = crickit.stepper_motor
# increase to slow down, decrease to speed up!
INTERSTEP_DELAY = 0.01

while False:
    print("Single step")
    for i in range(200):
        stepper_motor.onestep(direction=stepper.FORWARD)
        time.sleep(INTERSTEP_DELAY)
    for i in range(200):
        stepper_motor.onestep(direction=stepper.BACKWARD)
        time.sleep(INTERSTEP_DELAY)

    print("Double step")
    for i in range(200):
        stepper_motor.onestep(direction=stepper.FORWARD, style=stepper.DOUBLE)
        time.sleep(INTERSTEP_DELAY)
    for i in range(200):
        stepper_motor.onestep(direction=stepper.BACKWARD, style=stepper.DOUBLE)
        time.sleep(INTERSTEP_DELAY)
    print("Interleave step")
    for i in range(200):
        stepper_motor.onestep(direction=stepper.FORWARD, style=stepper.INTERLEAVE)
        time.sleep(INTERSTEP_DELAY)
    for i in range(200):
        stepper_motor.onestep(direction=stepper.BACKWARD, style=stepper.INTERLEAVE)
        time.sleep(INTERSTEP_DELAY)
[boot_out.txt](https://github.com/adafruit/circuitpython/files/3225230/boot_out.txt)

[crash.zip](https://github.com/adafruit/circuitpython/files/3225235/crash.zip)
`

dhalbert, 2019-05-28:
What version of CircuitPython are you using? `boot_out.txt` is unreadable, for some reason. Could you paste its contents here if it's readable to you?

[I edited your initial post to reformat the code: you can surround it in triple backquotes and then it formats nicely.]

jnalezny, 2019-05-28:
I am running adafruit-circuitpython-circuitplayground_express-en_US-4.0.1.uf2 that I just downloaded.  the libraries I got from adafruit-circuitpython-bundle-4.x-mpy-20190526.zip.  Both downloaded this afternoon.

jnalezny, 2019-05-28:
I managed to read this off the actual circuitplayground express:  
{ D 5 0 9 1 F D A - B B 2 8 - 4 C 5 3 - A 6 4 0 - 6 9 2 F D 9 8 8 0 C B 9 }   
if that helps

tannewt, 2019-06-25:
I've reproduced this and am looking at it now.

tannewt, 2019-06-25:
The cause is a corrupt mpy file (adafruit_crickit.py) that then breaks its code execution. I'm adding validation that catches this case but we should probably add a checksum.

kevinjwalters, 2019-06-25:
Was `adafruit_crickit.py` corrupt in the zip or did it get corrupted on its way to / on the CPX? If latter, do you know how that happened? Simple truncation or with changed bytes?

tannewt, 2019-06-25:
It's corrupt in the crash.zip above. It is the correct length but one block within the file is incorrect. I'm not sure how it happened.

kevinjwalters, 2019-06-25:
I've seen that hex unique id stuff in `boot_out` once. I think it was FAT corruption from Windows due to user error vs that FAT12 bug. Or possibly crashes, e.g. I have shorted CPX power pads twice and if stuff hasn't flushed the immediate crash that results won't be good as presumably host o/s doesn't know to flush or does it?

@jnalezny What host operating system are you running? Particularly for Windows, are you very methodical about always ejecting the `CIRCUITPY` drive before you unplug the CPX from USB?

kevinjwalters, 2019-06-26:
Does CircuitPython/MicroPython or the CP library build process use Microsoft **UUID** with that particular represesentaiton? My experience in that world is limited but I believe that's where I have seen that format string before. Is it used more widely than Microsoft libraries?

https://docs.microsoft.com/en-us/windows/desktop/rpc/generating-interface-uuids

commit 330517bd (2019-06-27): Validate portions of mpy load to detect corruption

Fixes #1917
closed 2019-06-27 by dhalbert