adafruit/circuitpython #1919: Expose a user-defined Frequency kwarg in BusIO's I2C

brentru, 2019-05-28:
Despite being a kwarg within the constructor (https://circuitpython.readthedocs.io/en/4.x/shared-bindings/busio/I2C.html#busio.I2C), Frequency is not a set-able property. 

Would it be possible to set the frequency via a call to `i2c.Frequency`. 

```
>>> i2c.
deinit          readfrom_into   scan            try_lock
unlock          writeto
```

tannewt, 2019-06-13:
Why do you need to change it as you go?

brentru, 2019-06-13:
@tannewt The ATECC608A requires the frequency to be decreased to wake the chip up, then increased again when a command needs to be sent.

tannewt, 2019-06-14:
Ugh, weird.

sommersoft, 2019-06-14:
> Ugh, weird.

Agreed. I looked at the sheet, and from what I can tell it's similar to a clock stretch, by holding SCL high for a minimum amount of time to wake from sleep.

Dropping the frequency does seem the best approach. I peeked at the M4 sheet and saw an SCL hold register/property, but that would likely be just as bleh to implement; if it would even address this. Not to mention far less useable across the ecosystem.

siddacious, 2020-04-06:
I ran into this indirectly while trying to use an MCP9600 because the MCP9600 library will fail with a cryptic "Input/Output Error" when you use `board.I2C` or otherwise use the default frequency.

It would be helpful for the library to be able to read the frequency in `__init__` and raise an exception if it isn't within an acceptable range (datasheet says 100K, examples use 200K apparently).

@cogliano your input would be appreciated

dhalbert, 2020-05-03:
Another example of why this would be useful is here: https://forums.adafruit.com/viewtopic.php?f=60&t=165077.

User wants to use the standard I2C bus for an external sensor that needs 100kHz instead of the standard 400kHz, but `board.I2C()` provides no way to set the frequency, and the CLUE library has already instantiated `board.I2C()`.

kevinjwalters, 2020-05-03:
Would this help those who are running the i2c bus over long cable lengths too?