adafruit/circuitpython #1922: wav sample_rate appears limited to less than 65536

kevinjwalters, 2019-06-04:
I wanted to experiement with some non audio playback and made some 8bit wavs to check playback. 64000 is ok but this one plays slowly and inspecting `sample_rate` shows why:

```
>>> wave_file = open("Evillaugh.96000.8bit.mono.wav", "rb")
>>> wave = audioio.WaveFile(wave_file)
>>> wave.sample_rate
30464
>>> 30464+65536
96000
```

I'd also be interested in any bottlenecks I'll encounter as I try faster sample_rate. Will CPX be capable to playing 100000 or 350000 8bit (or 16bit?) samples per sec from a wav file on the built-in flash?

C47D, 2019-06-04:
Appears to me it's because `sample_rate` is an `uint16_t` in the `audioio_wavefile_obj_t` type.

https://github.com/adafruit/circuitpython/blob/c5a4e19d6f76a09cba73149d5e3d4548da320899/shared-module/audioio/WaveFile.h#L34-L55

closed 2019-06-05 by tannewt

kevinjwalters, 2019-06-21:
Thanks @C47D, I tested 4.1.0-beta.0 last night with an 80kHz sample and this now appears to work fine: https://www.youtube.com/watch?v=WME0PRTvCk8

C47D, 2019-06-22:
Glad it worked, it does look beautiful, great music too. Is your project documented somewhere i would like to know more about it!

kevinjwalters, 2019-06-23:
I'm still debugging and cleaning code up but will let you know when it's out on t'Internet.

kevinjwalters, 2019-07-30:
Published last week: [Adafruit Learn: Making oscilloscope images with DACs](https://learn.adafruit.com/dac-oscilloscope-images)

PR #1926 (2019-06-05): [shared-module/audioio/WaveFile.h] Change sample_rate from uint16_t t…
…o uint32_t so it matches the sample rate type parsed from the WAV header format, fix #1922
commit 671178c8 (2019-06-05): [shared-module/audioio/WaveFile.h] Change sample_rate from uint16_t to uint32_t so it matches the sample rate type parsed from the WAV header format, fix #1922