adafruit/circuitpython #1923: Displayio / MiniTFTFeatherWing noise at left edge of screen

iraytrace, 2019-06-04:
The following code:
```
     import time
     from adafruit_featherwing import minitft_featherwing
     minitft = minitft_featherwing.MiniTFTFeatherWing()
     print('hello')
     while True:
         pass
```
Produces the following image with noise on the left side:
[image]

This noise persists even if the display is more substantially over-written as in:

```
import time
import board
from adafruit_featherwing import minitft_featherwing
import displayio
import terminalio
from adafruit_display_text import label

minitft = minitft_featherwing.MiniTFTFeatherWing()

label_count = 5
labels = list()
group = displayio.Group(max_size=label_count, scale=1, x=0, y=0)
font = terminalio.FONT
f_width, f_height = font.get_bounding_box()

colors = [0xffffff, 0xb0b0b0, 0xffff00, 0xb0b0b0, 0x00d0d0]
for i in range(label_count):
    labels.append(label.Label(font, max_glyphs=37, color=colors[i], line_spacing=0.94))
    labels[-1].x = 5
    labels[-1].y = i * f_height + int(f_height*.5)
    labels[-1].text = 'line %d' % i
    group.append(labels[-1])

minitft.display.show(group)
while True:
    pass
```

Which produces:

[image]

makermelissa, 2019-06-04:
Hi, which version of CircuitPython are you running this with? I tried it with 4.0.1 and it didn't produce this.

makermelissa, 2019-06-04:
I see on the forum, you grabbed an s3 build. I'm going to try compiling CP with the latest master to see if I can get this to reproduce.

makermelissa, 2019-06-04:
Ok, I was able to reproduce this with the latest build. It appears that the initial screen draw is not completely drawing everything.

tannewt, 2019-06-04:
@makermelissa I can take a look at this. I changed a good amount of this code so I'm not surprised there are bugs.

PR #1940 (2019-06-13): Add partial display update support.
Different operations to the display tree have different costs. Be
aware of these costs when optimizing your code.
* Changing tiles indices in a TileGrid will update an area
covering them all.
* Changing a palette will refresh every object that references it.
* Moving a TileGrid will update both where it was and where it moved to.
* Adding something to a Group will refresh each individual area it
covers.
* Removing things from a Group will refresh one area that covers all
previous locations. (Not separate areas like add.)
* Setting a new top level Group will  [... cut]
commit eb21fc3e (2019-06-13): Add partial display update support.

Different operations to the display tree have different costs. Be
aware of these costs when optimizing your code.
* Changing tiles indices in a TileGrid will update an area
covering them all.
* Changing a palette will refresh every object that references it.
* Moving a TileGrid will update both where it was and where it moved to.
* Adding something to a Group will refresh each individual area it
covers.
* Removing things from a Group will refresh one area that covers all
previous locations. (Not separate areas like add.)
* Setting a new top level Group will [... cut]
closed 2019-06-13 by dhalbert