adafruit/circuitpython #1930: [Wishlist] I2S support on feather nRF52840 express

C47D, 2019-06-08:
Hi,

I'm interested in audio applications using the nRF52840, as far as i know there is no support (yet) for the I2S peripheral on this device.

I did a very basic research and i think i know the big picture on how to add support for this and i would like to know if I am on the right track or if there is work already done in this regard, so i won't duplicate it.

So the nRF hal is based on nrfx which already have support for the I2S peripheral. I should implement `I2SOut.h` and `I2SOut.c` on `ports/nrf/common-hal/audiobusio`. The functions i should implement are the ones declared on the `shared-bindings/audiobusio/I2SOut.h`.

I was tacking a look at the atmel-samd implementation and i know it will be a hard task for a newcomer like me, but i think the acquired knowledge is worth the efford.

What do you all think?

sommersoft, 2019-06-08:
@C47D #1035 already has this as a goal.

> I did a very basic research and i think i know the big picture on how to add support for this and i would like to know if I am on the right track or if there is work already done in this regard, so i won't duplicate it.

I don't know of anyone working on it at the moment. I have `PDMIn` started, but its kind of stalled as of late. I will throw this out though: `AudioOut` also isn't done yet on the nRF52, which has proved a bit of a barrier to verify recording for me thus far. 

You are correct that `nrfx` is the HAL to use. And I wouldn't worry about the "newcomer" part. 

I'm going to close this one in favor of 1035, but I'll assign it to you.

closed 2019-06-08 by sommersoft

C47D, 2019-06-08:
Thanks @sommersoft

boutonce, 2020-05-02:
Any luck @C47D?  I'm very interested in this too.

C47D, 2020-05-02:
@boutonce hi, as far as I know there's support for I2S out on this board. Check the latest circuitpython.

boutonce, 2020-05-02:
@C47D - thank you. I did see a reference to it in CircuitPython, but is it supported in Arduino IDE? (I need interrupts and circuitpython doesn't support them currently to my understanding)

C47D, 2020-05-02:
I haven't used CP in a while, I think is better if you open a new issue because this one is too old.