adafruit/circuitpython #1935: GamePadShift requires a delay

makermelissa, 2019-06-11:
One of the things that I found while working with GamePadShift is that reading it too fast will sometimes return 0 even if there are buttons pressed. I believe this is because it shifts the bits out when it is read and doesn't maintain state. I'm not sure if this was the intended behavior or just a result of a fast CPU reading faster than the shift register is reading the buttons. I think it would be useful to maintain the state after reading the buttons so subsequent reads don't show up as no buttons being pressed.

ladyada, 2019-06-11:
yeah @deshipu mentioned this before i think? it would be nice if we had a queue of button presses!

deshipu, 2019-06-11:
It's not a bug, this is how it is supposed to work, and how it is documented. Reading the buttons state clears the buffer, and it will remain empty until new button presses are registered. This is shouldn't be a problem in any normal application, as the buttons are being read every 30ms or so, and no human is able to see that.
I suppose it can be surprising if you read the button state in a tight loop more than once per frame, but why would you ever need to do that?

deshipu, 2019-06-11:
We could make it work as expected if instead of clearing the buffer, we set a flag that tells the code that updates the buttons to clear the buffer when it reads their state. But I don't think it's worth the effort without an actual use case that would require it.

makermelissa, 2019-06-11:
The use case is @kattni is working on a library to make accessing buttons easier and is running into them returning 0 sometimes when they're still being pressed.

deshipu, 2019-06-11:
Is there code available?

makermelissa, 2019-06-11:
She'll have to give that to you. However, I had to add a delay for the same reason in https://learn.adafruit.com/pybadge-conference-badge-multi-language-unicode-fonts

kattni, 2019-06-11:
It's a sparse WIP. I can create a quick repo and push the code if you want it. It's designed to work on PyGamer/PyBadge, though only tested on gamer so far. 

I'm writing a library that's meant to be designed like the Circuit Playground Express `cpx` library - a general purpose wrapper library that makes all the functionality of the boards available in one place. The first thing I did to test the button code was the equivalent of `print(pybadger.get_pressed())` in a `while True:` loop. It didn't occur to me that there would be any issue with it. I guess there's very little use case that wouldn't have other code in the loop or have the buttons doing something which would introduce a delay anyway, but it's not out of the question for a beginner user to do what I did as a first attempt to begin to understand button presses.

deshipu, 2019-06-11:
I will try to propose a pull request that lets you use it like that.

PR #1939 (2019-06-12): When clearing gamepad buffer, use the last button state, not 0
When reading the accumulated button presses in gamepad and gamepadshift,
don't clear the buffer to "no buttons pressed", but instead set it to
the current (last checked) state. This clears the accumulated presses,
but retains any ongoing ones.

This fixes #1935
commit 5e265f4f (2019-06-12): When clearing gamepad buffer, use the last button state, not 0

When reading the accumulated button presses in gamepad and gamepadshift,
don't clear the buffer to "no buttons pressed", but instead set it to
the current (last checked) state. This clears the accumulated presses,
but retains any ongoing ones.

This fixes #1935
closed 2019-06-13 by tannewt