adafruit/circuitpython #1941: Adafruit SI1145 UV Lux sensor missing libraries for Raspberry Pi in python

kflmiami420, 2019-06-13:
I am working on a a personal project that needs Uv Index I purchased the SI1145 for my raspberry pi .   There are no Adafruit_CircuitPython_SI1145 libraries,drivers in python for this sensor to be used with a raspberry pi. 

Please create a driver so I can get UV Index readings from this sensor

ladyada, 2019-06-13:
hiya that sensor isn't recommended for UV readings, as it does not have a true UV sensor. try the VEML7700!

caternuson, 2019-06-14:
So is this a yay or nay for having a driver?

Related:
https://forums.adafruit.com/viewtopic.php?f=8&t=152822

ladyada, 2019-06-16:
ya, thats the same person. i think this is a very low priority driver, but if someone wants to do it, that would be very welcome :) they may benefit from checking the silabs fm transmitter, the 'i2c data access technique' silabs uses is similar

gogreen53, 2020-07-16:
Adding a +1 request for a library since I just bought one of these the other day. I'd at least like to read IR values off it.

socraticmethod00, 2022-02-23:
I'm a bit late to the SI1145 party but was also very disappointed Adafruit did not have a circuitpython driver for a sensor they are currently  selling.  I have created my first Raspberry Pi Pico project with an attempt at the circuitpython driver for the SI1145.   Just opened my github account , here's the link:  https://github.com/socraticmethod00/pico

DeadSix27, 2022-06-08:
Any update on this? [socraticmethod00](https://github.com/socraticmethod00)'s versions a good start though

caternuson, 2022-07-19:
Another request:
https://forums.adafruit.com/viewtopic.php?f=60&t=192880

caternuson, 2022-07-20:
Worked up a basic library here:
https://github.com/caternuson/Adafruit_CircuitPython_SI1145

caternuson, 2022-07-26:
Repo moved to Adafruit:
https://github.com/adafruit/Adafruit_CircuitPython_SI1145
and PR made to add to bundle:
https://github.com/adafruit/Adafruit_CircuitPython_Bundle/pull/388

tekktrik, 2022-09-16:
@caternuson Think this is good to close since the repo exists?

caternuson, 2022-09-19:
Yep, thanks. Closing.

closed 2022-09-19 by caternuson