adafruit/circuitpython #1944: 4.1.0-beta.0 terminalio with ili9341 no longer uses whole screen

anecdata, 2019-06-14:
Adafruit Feather M4 Express with samd51j19 (31b)
CircuitPython 4.1.0-beta.0 on 2019-06-13
CircuitPython Library Bundle 20190613
2.4" TFT FeatherWing

```"""
This test will initialize the display using displayio
and draw a solid red background
"""

import board
import displayio
import adafruit_ili9341
import time

spi = board.SPI()
tft_cs = board.D9
tft_dc = board.D10

displayio.release_displays()
display_bus = displayio.FourWire(spi, command=tft_dc, chip_select=tft_cs)

display = adafruit_ili9341.ILI9341(display_bus, width=320, height=240)

# Make the display context
splash = displayio.Group(max_size=10)
display.show(splash)

color_bitmap = displayio.Bitmap(320, 240, 1)
color_palette = displayio.Palette(1)
color_palette[0] = 0xFF0000

bg_sprite = displayio.TileGrid(color_bitmap,
                               pixel_shader=color_palette,
                               x=0, y=0)
splash.append(bg_sprite)

time.sleep(1)
display.show(None)
while True:
    time.sleep(1)
    print("Test")
    pass```

With minor changes to the library example, text only scrolls at the top of the screen, leaving the bottom blank.

tannewt, 2019-06-14:
Please post a picture.

anecdata, 2019-06-14:
For comparison, hx8357 with 3.5" FeatherWing works fine in this config.

anecdata, 2019-06-14:
Added a loop cointer so it's clear the print statements are unique. Some code uses a little more of the top of the screen than others, but never more than half, that I've seen.
[image]

anecdata, 2019-06-14:
Well, now that I'm banging on it more, it seems that maybe the init just got in a funny state. I'm able to get the whole screen now. Successive reloads didn't clear something, had to remove battery and completely power down. Thought I had done that last night, but possibly not. I'll close this if I don't see any strangeness in the next couple of days.

makermelissa, 2019-06-14:
Yeah, sometimes when working with displays they'll stick with the first init codes they see and if you don't unplug completely this can either lull you into a false sense of security or see issues where there aren't any. I've had it happen to me a few times while working on the drivers. :)

makermelissa, 2019-06-14:
Perhaps the display isn't resetting properly.

anecdata, 2019-06-14:
It's possible my display is quirky, I don't have a 2nd to test. But I put 4.0.1 back on and it worked fine. But 4.1.0-beta.0 is still acting up after that.

makermelissa, 2019-06-14:
I'm thinking maybe partial display updates exposed something. Do you happen to have a PyPortal to test with? It also uses the ili9341?

anecdata, 2019-06-14:
PyPortal terminalio is fine, but I believe it uses parallel interface to TFT, not SPI.

makermelissa, 2019-06-14:
Yeah, but at least it's a good data point. I can hook up a display in a little bit and see if the behavior is the same on mine. I also have a few different ili9341 displays.

dhalbert, 2019-06-14:
Or maybe the new interpreter speedup means it needs a delay now somewhere in the Python code?

anecdata, 2019-06-14:
Oddly (though perhaps related), I'm not seeing discernible speedup on the 2.4" FeatherWing.

makermelissa, 2019-06-14:
Ok, here's what I observed while testing it using the above code. When you start it, it paints a  red rectangle on the entire screen fine and then prints test starting from the bottom. 

Each time you save the file it repaints the red rectangle just fine showing that the init commands seem fine. However, the terminal seems to shrink a little each time. So this may be related to terminalio.

anecdata, 2019-06-17:
MiniTFT FeatherWing exhibits the same display bahevior: graphics are fine, terminalio loses about one text line per reload.

anecdata, 2019-06-17:
Some updates to above comments:

The 2.4" FeatherWing does appear to run faster with 4.1.0-beta.0 fresh out of a power cycle.

But now working on the 3.5" FeatherWing more, I am seeing some quirky behavior. It's in a super slow mode now that I can't get out of even through reverting to 4.0.1 (or back) and / or power cycling. Could be specific to my unit (I only have the one to test), or it possibly implies there is some static storage on the display chip that isn't getting completely reset on powerup. Perhaps related... even if I try to speed up SPI from default 250000 to something higher, after a short time, `spi.frequency` will show 250000 again.

Gist here: [https://gist.github.com/anecdata/771c7ad55e41fd1baa1b846d6994b0f9](https://gist.github.com/anecdata/771c7ad55e41fd1baa1b846d6994b0f9)

anecdata, 2019-06-17:
More testing in the Gist comments. Observations:

- "slow mode" seems independent of whether an external display is actually connected

- therefore, seems to be tied to processor / circuitpython rather than the external display hardware

- hardware reset button during run seems to be a trigger for slow mode

- slow mode persists across reloads

- SPI frequency is sticky across reloads in normal mode, not in slow mode

- normal mode can be re-established by reloading without display-specific code

tannewt, 2019-06-26:
@anecdata can you try the latest S3 build? I made some tweaks for SPI displays in it.

anecdata, 2019-06-26:
Still seeing successive loss of lower display lines on successive reloads using code above on:

Adafruit Feather M4 Express with samd51j19
adafruit-circuitpython-feather_m4_express-en_US-20190625-d3adfde.uf2
CircuitPython Library Bundle 20190625
2.4" TFT FeatherWing (ili9341)

Default SPI speed appears to have been raised from 250000Hx to 24MHz.

PR #1974 (2019-06-27): Properly reset the terminal each init.
Fixes #1944
commit 2004be96 (2019-06-27): Properly reset the terminal each init.

Fixes #1944
closed 2019-06-27 by tannewt