adafruit/circuitpython #1947: [pygamer] IOError when reading files from SD card

shazz, 2019-06-15:
Hi,
I just figure out that there is a 'slight' difference when reading files from the flash or from the SD card.

**Version**: master (6/16/2019)

**File generation** (copied to the flash and to the SD card
````bash
time dd if=/dev/urandom of=test.dat bs=1 count=1M
````

**Code**:
````python
import sys
import os
import adafruit_sdcard
import board
import busio
import digitalio
import storage

# Connect to the card and mount the filesystem.
spi = busio.SPI(board.SCK, board.MOSI, board.MISO)
cs = digitalio.DigitalInOut(board.SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")
sys.path.append("/sd")
os.chdir("/sd")

buf = bytearray(160*128)
f = open('test.dat', 'rb')
f.readinto(buf)
f.close()
````

**Results on the SD card:**
````
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] Input/output error
````

works well from the flash.
`os.listdir()` works, files located on the SD card are shown

Any idea when I can check ? Libraries ?

jerryneedell, 2019-06-15:
When I see this error ```Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
OSError: [Errno 5] Input/output error
```
it is a problem reading the program from the Files system -- not related to the actual code or hardware.
try copying your program over again and making sure the file system has been properly synched.

shazz, 2019-06-15:
In fact I ran the test code:

- from the flash
- from the SD card
- from the REPL (line by line)

same result.
I also reformat the SD card. No change.
Running script from the SD card works, but that's really this `readinto()` which fails.

shazz, 2019-06-15:
Ah, looks it comes from here in `extmod/vfs_fat_file.c`

````C
STATIC mp_uint_t file_obj_read(mp_obj_t self_in, void *buf, mp_uint_t size, int *errcode) {
    pyb_file_obj_t *self = MP_OBJ_TO_PTR(self_in);
    UINT sz_out;
    FRESULT res = f_read(&self->fp, buf, size, &sz_out);
    if (res != FR_OK) {
        *errcode = fresult_to_errno_table[res];
        return MP_STREAM_ERROR;
    }
    return sz_out;
}
````

then....

ladyada, 2019-06-15:
maybe same issue as
https://github.com/adafruit/Adafruit_CircuitPython_SD/issues/14

shazz, 2019-06-15:
There is maybe a link... in this case **read** **crashes** after **200** bytes (versus **hangs** when **write** more than **1023** bytes).

Bu in any case, there is an issue with reading and writing on the SD card...

dhalbert, 2019-06-16:
What is the brand, model, and size of the card? Some cards work worse than others.

We have a C-based library we are planning to use instead of the python-based SD library.

dhalbert, 2019-06-16:
I should say _thinking_ of substituting, since we may not have space on some boards.

shazz, 2019-06-16:
That's a SanDisk 8GB micro SDHC class 4. 
I will try with some others...

shazz, 2019-06-21:
So I tested on 3 micro SD cards:

| SD Card       | class | type | size   | Read speed | Access time | Results |
|---------------|-------|------|--------|------------|-------------|---------|
| Sandisk       | 4     | SDHC | 8GB    | 46.8 MB/s  | 0.53ms      | Failed  |
| Team          | 4     | SDHC | 8GB    | 22.5MB/s   | 1.48ms      | Success |
| Sandisk ultra | 10    | SDHx | 128 GB | 86.2 MB/s  | 0.45ms      | Failed  |
| Sandisk ultra | 10    | SDHC | 8GB    | 45.9MB/s   | 0.69ms      | Failed  |

I did not expect the cheap slow card to be the only working :D

shazz, 2019-06-21:
For reference I did some performance tests between the pygamer (120MHz) and my STM32 based board running MicroPython at 84MHz:

| method   | size  | time ms  | time in ms (reference) |
|----------|-------|----------|------------------------|
| readinto | 10    | 0.250244 | 0.02592                |
| readinto | 100   | 1.15051  | 0.08985001             |
| readinto | 1000  | 10.0525  | 0.749                  |
| readinto | 10000 | 76.3     | 1.94809                |
| readinto | 10240 | 75.9979  | 2.45233                |
| readinto | 20480 | 150.351  | 5.75184                |
| readinto | 40960 | 302.096  | 6.57845                |

So... yes... I vote for the C-based library!

kevinjwalters, 2019-07-03:
BTW, I asked about SDXC support as a fellow user had found a large SD card didn't work: https://forums.adafruit.com/viewtopic.php?f=19&t=152188&p=751225&hilit=sdxc#p751225

Anton-2, 2019-10-14:
Any information about replacing the python-based SD library by a C version ?

In the meantime, you can accelerate things by fixing https://github.com/adafruit/Adafruit_CircuitPython_SD/issues/22 .

Anton-2, 2019-10-19:
There is a problem with fast/recent cards when reading multiple blocks.
Basic fat operations will work (as they access just one block at once), and so for any reads less than 1024 bytes long.

But reading two or more blocks will (sometimes) get a return value of 127 or 63, and data will be mangled.

See https://github.com/adafruit/Adafruit_CircuitPython_SD/issues/25.

dhalbert, 2019-10-31:
@shazz could you re-test with the latest version of https://github.com/adafruit/Adafruit_CircuitPython_SD ? Thanks.

dhalbert, 2020-02-20:
This should be fixed by https://github.com/adafruit/Adafruit_CircuitPython_SD/pull/26. Closing for now. Please start a new issue in https://github.com/adafruit/Adafruit_CircuitPython_SD if you still see SD problems. Thanks.

closed 2020-02-20 by dhalbert