adafruit/circuitpython #1960: [displayio] trapezoid/triangle primitive

ladyada, 2019-06-19:
this is what graphics engines tend to do. we can have a C-optimized triangle or trapezoid drawing program and everything else uses it (polygons/lines/shapes). outline and fill.
maybe also a circle helper at some point but this is more...primitive! ha.

tannewt, 2019-07-01:
I was trying to avoid having the line algorithms in the C code natively and rely on Shape to store solid colors efficiently.

How would triangle primitives be used and why would they need native speed?

ladyada, 2019-07-01:
i bumped into this when looking at how to draw 'thick' lines, all the algorithms ended up saying "use a trapezoid primitive' - also if/when we do filled triangles or polygons, they're slow to do compared to rect-fills

tannewt, 2019-07-01:
Mind linking me to examples? I'm new to this graphics stuff too. :-)

ladyada, 2019-07-01:
this is the best document on how to draw thick lines.primitives. https://stackoverflow.com/questions/1222713/how-do-i-create-a-line-of-arbitrary-thickness-using-bresenham

ladyada, 2020-05-13:
got added in `vectorio`!

closed 2020-05-13 by ladyada