adafruit/circuitpython #1965: Python version equivalent

nikmolnar, 2019-06-21:
Where can I find which version of Python that CircuitPython implements? I've seen references elsewhere to Python 3.4, but I can't seem to find any official record of this.

Also, I assume CircuitPython will implement newer versions over time, in which case it will become important to know which versions of CircuitPython implement which Python versions.

Thanks!

dhalbert, 2019-06-21:
CircuitPython is forked from MicroPython, and takes its core language interpreter essentially intact. MicroPython has stated that it is currently based on core features in CPython 3.4. Some features are omitted or are slightly different frp, C{utujpm to save time or space. There's now discussion of what to add to MicroPython for the next major release. For background see these MicroPython issues:

https://github.com/micropython/micropython/issues/1329
https://github.com/micropython/micropython/issues/2415
https://github.com/micropython/micropython/issues/4821

Note that "support for Python 3.4" does not include the library, only the core language.

We merge from upstream MicroPython infrequently, to pick up bug fixes and feature additions. We've done it in a major way twice so far, since the fork.  These merges take significant time to do, so it's hard to do them frequently. We will continue to merge from upstream, at points when the MicroPython release is considered fairly stable, and when it's worth doing the merge to get significant changes.

nikmolnar, 2019-06-21:
Thank you @dhalbert , that's exactly what I wanted to know!

closed 2019-06-21 by nikmolnar