adafruit/circuitpython #1966: Issue running on the makerdairy nrf52840-mdk

JJJollyjim, 2019-06-22:
When I flash the latest build (either my own build of master, or [your provided one](https://adafruit-circuit-python.s3.amazonaws.com/bin/makerdiary_nrf52840_mdk/adafruit-circuitpython-makerdiary_nrf52840_mdk-en_US-4.0.0-alpha.5.hex) on an erased nrf52840-mdk (non-dongle version), the chip produces no serial output. This happens whether I flash with the makefile (via pyOCD), DAPLINK, or gdb.

Debugging in gdb, the code seems to begin in `cpu_get_regs_and_sp`, which it runs through, until the `bx` instruction jumps it to the address 0xfffffffc (the offset for the relative jump is -40), and it hangs forever.

I suspect the code isn't supposed to start in that function... is there maybe a linker script bug or something that's causing that?

JJJollyjim, 2019-06-22:
(My only deviation from the https://learn.adafruit.com/circuitpython-on-the-nrf52/ guide was that I didn't run the `./bluetooth/download_ble_stack.sh` script, since it appears not to exist...)

jerryneedell, 2019-06-22:
If I recall correctly - the bluetooth stack no longer needs to be downloaded separately.
FWIW - I have been able to build/ load CP for the makerdary nrf52840 usb dongle, but one difference is tat I installed the UF2 Bootloader to stand use that to load CP.

JJJollyjim, 2019-06-22:
Hmm it looks like the UF2 bootloader isn't supported on the non-dongle (since it comes with its own programming hardware)

JJJollyjim, 2019-06-22:
Heck, I didn't have a softdevice flashed. Nevermind! :)

closed 2019-06-22 by JJJollyjim