adafruit/circuitpython #1967: PyGamer safe mode crash into Hardfault_Handler

kattni, 2019-06-24:
Working on Adafruit_CircuitPython_PyBadger. At that state of the lib, if I moved the order of things in code.py, it would:

1. Reboot the board.
2. Returns `OSError 5: Input/Output` (not sure past this, it flashes onto the display, so I'm only seeing what fits on to the display)
3. Reboots again into safe mode with `Hardfault_Handler`.

If the `pybadger.badge` line is inside the `while True:` loop, the code runs fine. If I move it above the loop, safe mode.

Including zip containing the working state of Adafruit_CircuitPython_PyBadger at the time of the error, the safe mode state code.py, and the working code.py.

[pybadger_safe_mode_code.zip](https://github.com/adafruit/circuitpython/files/3322355/pybadger_safe_mode_code.zip)

tannewt, 2019-06-25:
I tried to reproduce this but was unable to with a build of 4.0.x latest and with 4.1.0-beta.0. What version were you using? Can you post your full drive? I had to pull down a number of libraries to get it to run.

kattni, 2019-06-25:
I'm using `Adafruit CircuitPython 4.0.1 on 2019-05-22`. I tested it again and it failed to safe mode. Included is the current state that failed to safe mode.

[pygamer_safe_mode_drive.zip](https://github.com/adafruit/circuitpython/files/3327032/pygamer_safe_mode_drive.zip)

tannewt, 2019-06-25:
This is fixed in 4.0.x. I can repro on 4.0.1 but not 4.0.x. It's the mistaken free of the root Group causing issues which was fixed in #1928. Will release 4.0.2 in the next 24 hours.

closed 2019-06-25 by tannewt