adafruit/circuitpython #1970: Update shared-bindings/index.rst

dhalbert, 2019-06-25:
The list of supported modules per port is out of date in a number of ways.

It would be nice to have this generated automatically, but that might not be easy.

sommersoft, 2019-06-25:
> It would be nice to have this generated automatically, but that might not be easy.

An **"immediate and unverified"** thought would be to have a python script that scans the base definitions (`circuitpy_mpconfig.mk`, `shared-bindings`, etc), then cycles through each board's `mpconfig.mk` matching against the base. Output is compiled into a JSON file. Then, on the Sphinx side, have a Jinja template that builds the `.rst` page.

Quick related searches:
- https://www.ericholscher.com/blog/2016/jul/25/integrating-jinja-rst-sphinx/
- https://www.sphinx-doc.org/en/master/templating.html#do-i-need-to-use-sphinx-s-templates-to-produce-html

sommersoft, 2019-07-30:
While this is "initially" done, there is a slight problem. Which actually isn't new to us.

The generated table does not wrap the text: https://circuitpython.readthedocs.io/en/latest/shared-bindings/support_matrix.html#support-matrix

After some searching, this is primarily a `sphinx-rtd-theme` [issue](https://github.com/readthedocs/sphinx_rtd_theme/pull/432).

There exists a workaround, which is already implemented in `docs/static/customstyle.css`. However, it doesn't seem to work for this particular table. That may be related to how the `rstjinja` extension is pushing the output to the html builder. 

Do we want to leave this issue open, or create a new one?

tannewt, 2019-07-30:
I think this is ok to close. It's probably best to have separate pages for each board which explain what is present and what is missing.

sommersoft, 2019-07-31:
Closing, then. I'll make a new (or two) issue with some outlying issues, and include the "future page-per-board".

closed 2019-07-31 by sommersoft

dhalbert, 2019-08-13:
For your new issues: table looks nice informationally but doesn't wrap;

[image]

sommersoft, 2019-08-13:
Yep. I haven't been actively working it, but I opened #2020 to track it.