adafruit/circuitpython #1976: Enabling MICROPY_PY_DELATTR_SETATTR for SAMD51

arturo182, 2019-06-28:
Has anyone looked at the cost of enabling `MICROPY_PY_DELATTR_SETATTR`?

It's off by default, but it's on for the stm32 port, and I just found out it was off the hard way, by trying to debug why my code doesn't work for 2h :<

According to the commit enabling it in stm32, there's no performance hit in the classes that don't implement it, see https://github.com/adafruit/circuitpython/commit/190c7dba89dbb1165f682b56f278e3bc9715680c

The commit also mentions `MICROPY_PY_BUILTINS_NOTIMPLEMENTED` which is enabled in SAMD51 builds already.

If we could enable it for SAMD51, that would help me a lot :pray:

arturo182, 2019-06-28:
Here's a build straight from master:
```
234496 bytes free in flash out of 499712 bytes ( 488.0 kb ).
178776 bytes free in ram for stack out of 196608 bytes ( 192.0 kb ).
```

And here's a build with `MICROPY_PY_DELATTR_SETATTR=1`:
```
234376 bytes free in flash out of 499712 bytes ( 488.0 kb ).
178776 bytes free in ram for stack out of 196608 bytes ( 192.0 kb ).
```

To sum up:
-120B free flash
-0B free stack

This was for `feather_m4_express`.

tannewt, 2019-07-01:
Did you try it for all builds? It'd be awesome if it was uniformly on. (Please make a PR for it too.) Thanks!

dhalbert, 2023-10-28:
This got turned out at some point. I tested on Trinket M0, Metro M4, RP2040.

dhalbert, 2023-10-28:
Sorry, this is about `__setattr__`() and `__delattr__()`, not `setattr()` and `delattr()`. Reopening.