adafruit/circuitpython #1977: Metro M4 Express additional UART handling incorrect

rce1086, 2019-06-29:
### Metro M4 Express
UART on pins 0/1 used for serial LCD display: attempting to create an additional UART (needed because of no raw serial stream being available from host PC via USB) on pins D10,D11 (as indicated by the example script as a usable pair)
`ascom = busio.UART(
    board.D10,
    board.D11,
    baudrate=9600,
    bits=8,
    parity=None,
    stop=1,
    timeout=0.25,
    receiver_buffer_size=64)` 
doesn't work - reception on D11 is OK, but nothing gets send on D10 (as shown by logic analyser on pins); experimenting with other pins (using D13 for TX, for example, with RX on D11) does work, as do A4/A5, suggesting that only pad0 can successfully be used for TX
The CircuitPython sources are written to assign either pad0 or pad1 (via a division on TXPO), but it appears that only pad0 assignments work...

ladyada, 2019-06-29:
thats correct, on SAMD51/M4 boards TX can only be pad 0, RX can be any pin

[image]

rce1086, 2019-06-29:
Thanks for confirming this, Lady Ada.
May I request, therefore, that the CircuitPython code be tightened up to reject these inappropriate settings, rather than setting configurations which won't work, thus making the output of the sample script [https://github.com/adafruit/Adafruit_Learning_System_Guides/blob/master/CircuitPython_Essentials/UART_Test_Script.py](url)
reflect reality...
Thanks to all involved, I'm liking CircuitPython :-)

ladyada, 2019-06-29:
yep we'll look at it - thanks for the bug report :)

tannewt, 2019-07-01:
@rce1086 Would you like to take a stab at fixing this? You found the right place. That test needs an ifdef to correct it for SAMD51 similar to what is done here: https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/common-hal/audioio/AudioOut.c#L124

rce1086, 2019-07-05:
Sorry for the delay in replying: yes, I will - I assume you're looking for a pull request?

tannewt, 2019-07-05:
Yes please!

tannewt, 2019-07-15:
Fixed in #1989. Thank you!

closed 2019-07-15 by tannewt