adafruit/circuitpython #1980: import multiterminal throws ImportError

iot49, 2019-06-30:
```python
>>> import multiterminal

Traceback (most recent call last):
  File "<stdin>", line 9, in <module>
ImportError: no module named 'multiterminal'
```

I tried UNSUCCESSFULLY adding defines to circuitpy_mpconfig.mk.

```
ifndef CIRCUITPY_MULTITERMINAL
CIRCUITPY_MULTITERMINAL = 1
endif
CFLAGS += -DCIRCUITPY_MULTITERMINAL=$(CIRCUITPY_MULTITERMINAL)
```

and circuitpy_defns.mk

```
ifeq ($(CIRCUITPY_MULTITERMINAL),1)
SRC_PATTERNS += multiterminal/%
endif
```

tannewt, 2019-07-01:
What are you trying to do? multiterminal was only ever implemented for ESP8266 which is no longer supported.

iot49, 2019-07-01:
I’m trying to catch <stdout> and <stdin> (specifically while executing
“exec()”).

(Assigning sys.stdout is not supported in micro/circuitpython).

Bernhard

iot49, 2019-07-05:
I solved the "problem" with a handler function in `supervisor/shared/micropython.c` that, when set, is called with whatever is written to serial. Handles output only and is much simpler than multiterminal.

Contact me if interested.

closed 2019-07-05 by iot49