adafruit/circuitpython #1994: CircuitPython json.dumps support for JSON Pretty-Print output

yesnoio, 2019-07-13:
Hoping to propose a small revision to CircuitPython's `json.dumps` so it supports pretty-printing of JSON strings. It seems `ujson.dumps` supports two additional formatting arguments: `json.dumps(parsed, indent=4, sort_keys=True)` (https://stackoverflow.com/a/12944035/6763539).
Some may want this pretty-print functionality, while others may find it unnecessary. Given how Python supports import of specific module functions, perhaps there should be `json.dump` & a `json.dumps` where `json.dumps` is status-quo & `json.dump` prints JSON in a pretty format?

xstook, 2023-05-03:
Just ran into this today, took me awhile to realize it's not supported in CircuitPython. This feature would be much appreciated!

joecorsi, 2025-05-22:
same here, woud love some indents.