adafruit/circuitpython #2001: Consider using / convert u8g2 for displays instead of SSD1306 only

NightSkySK, 2019-07-19:
I've noticed that for CircuitPython the is a quite limited number of displays supported. 
Please consider to use/convert to CircuitPython U8G2 library from https://github.com/olikraus/u8g2 It's just awesome.

tannewt, 2019-07-19:
The model of that library is completely different from our model. In CircuitPython we use Python files to store init sequences while u8g2 uses C. Using u8g2 would bloat our binaries code size a ton.

We could use it as a reference for creating community drivers for all of the different chips though.

closed 2019-07-19 by tannewt

NikilXYZ, 2022-01-14:
Are there drivers now for SSD11327 chip?

ladyada, 2022-01-14:
https://learn.adafruit.com/adafruit-monochrome-1-12-in-128x128-oled