adafruit/circuitpython #2008: Using WIZNET5K without attached ethernet causes hang

brentru, 2019-07-23:
Using an Adafruit Ethernet FeatherWing with a Feather M4 Express, without an ethernet cable attached causes the REPL to become non-responsive. The second time, after resetting the board, it caused a crash and the repl printed that it crashed into the _HardFault_Handler_.

The feather is currently unable to switch back into CircuitPython, only the bootloader. 

Code:
```
import board
import busio
import wiznet
import socket
import time
spi = busio.SPI(clock=board.SCK, MOSI=board.MOSI, MISO=board.MISO)
eth = wiznet.WIZNET5K(spi, board.D10)
while eth.ifconfig()[0] == '0.0.0.0':
    time.sleep(1)

host = 'example.com'
port = 80
fam, typ, pro, nam, socketaddr = socket.getaddrinfo(host, port)[0]

for x in range(0,10):
    ss = socket.socket(fam, typ, pro)
    ss.connect(socketaddr)
    ss.send('GET / HTTP/1.0\nHost: %s\n\n' % host)
    print(ss.recv(1000))
    ss.close()
```

Problematic portion of the code:
```
eth = wiznet.WIZNET5K(spi, board.D10)
while eth.ifconfig()[0] == '0.0.0.0':
    time.sleep(1)
```

Possibly the Wiznet module should return an OSError or something similar if it's unable to get an address by DCHP.

nickzoic, 2019-08-01:
So the problem is that the DHCP process is run asynchronously: the python code keeps running while the wiznet device is attempting the get an address.  If it can't get an address then the while loop will never exit: you'd need to do something cleverer like giving up after 20 times around the loop and trying something else.

*However*, you should be able to interrupt the `time.sleep(1)` in the REPL (with ctrl-C) and the hard fault obviously shouldn't be happening and the flash shouldn't have been corrupted!

Can you let us know the exact circuitpython version you were using, eg: the filename you flashed it from or the string printed when you reset the board?

brentru, 2019-08-01:
@nickzoic  Would it be possible to have a `wiznet.connected` property within the wiznet module which I would poll against, instead of the while/sleep code?

> . If it can't get an address then the while loop will never exit: you'd need to do something cleverer like giving up after 20 times around the loop and trying something else.

Should this be handled within the wiznet module itself? Or in the code.

> Can you let us know the exact circuitpython version you were using, eg: the filename you flashed it from or the string printed when you reset the board?

Was using a Feather M4 Express on 4.0.2 stable. I do not have the exact ver-str since it was corrupted/couldn't recover.

nickzoic, 2019-08-01:
4.0.2 stable is enough ... I'll try to replicate the problem on the weekend.
Have you been able to re-flash the board since?

I'm not sitting in front of the hardware right now, but there's a read-only `connected` property on the interface itself, eg:
```
eth = wiznet.WIZNET5K(spi, board.D10)
while not eth.connected:
    print("plug me in!")
    time.sleep(1)
```
You still have to poll it though.  Also, `connected` just tells you if there's a cable to a switch port: you still have to check `ifconfig` in case there's no DHCP server on the network, or the DHCP hasn't succeeded yet.

> Should this be handled within the wiznet module itself? Or in the code.

At the moment the DHCP process happens asynchronously, so your python code can get on with other things while it is working[*] ... but the downside of this is that your code has to do the error handling.  Arguably, given CircuitPython's focus on simplicity for the user, it might be better to block until an address is allocated and throw an exception on timeout.  There's a `dhcp` kwarg on the constructor to activate or deactivate dhcp, we could add another kwarg to switch to this synchronous behaviour.  What do you reckon, @tannewt ?

[*] DHCP can take a few seconds to work even in the best case: this is annoying but it's what we're stuck with in IPv4.  For IPv6 see RFC4429 :-)

brentru, 2019-08-02:
> You still have to poll it though. Also, connected just tells you if there's a cable to a switch port:

Ok, great! EthernetManager would take care of that.

When you set up this weekend, could you try checking to see if ifconfig succeeded, without plugging in the ethernet cable? It _should_ cause a hardfault.

> Have you been able to re-flash the board since?

Yup - had to run SPIFlash on it first. Adding a UF2 to FEATHERBOOT couldn't recover it.

>  Arguably, given CircuitPython's focus on simplicity for the user, it might be better to block until an address is allocated and throw an exception on timeout.

I like this method, the `timeout` should be user-configurable.

nickzoic, 2019-08-04:
* Can't replicate on ~4.0.0 e00a702ce (which is what was still on my board) ... Ctrl-C breaks out of the loop as expected.
* Ditto a fresh build of 4.0.2 ... I can run your code above w/ no cable attached & it breaks out fine.  DHCP discovery is running happily in the background.
* eth.connect works as expected ... it depends on the physical interface, not the network actually being useful.
* when DHCP succeeds, the loops exits as expected and example.com is coming through fine.

In short, I'm a bit puzzled.

nickzoic, 2019-08-06:
Hmmm, short of the usual checks (is your power supply good enough? Is the 'wing seated properly? Does it still do it with the latest release? Can we try over with new hardware?) I'm stumped.  One thing you could try is to plug an ethernet cable into a port you can monitor and check that you can see DHCP discovery packets being sent while waiting for ifconfig() to finish, but I'm clutching at straws here.

nickzoic, 2019-08-06:
tried it again with no hardware reset line attached (because that's possible) and after a soft reset of the board (which happens) but still all working as expected.

nickzoic, 2019-08-28:
No worries.