adafruit/circuitpython #2013: Replace most uses of MICROPY_VM_HOOK_LOOP with RUN_BACKGROUND_TASKS

dhalbert, 2019-07-26:
`MICROPY_VM_HOOK_LOOP` was provided in MicroPython `vm.c` to allow tasks to be run at safe points (between bytecodes) in the VM. It is defined in CircuitPython as `run_background_tasks();`, and we use it many more places when we're busy waiting for an indefinite period and need to service background tasks. It's also guarded by `#ifdef`, rather unnecessarily.

Let's create a new macro and redefine the old:
```
#define RUN_BACKGROUND_TASKS run_background_tasks()
...
#define MICROPY_VM_HOOK_LOOP RUN_BACKGROUND_TASKS;
```
Then change nearly all the uses of `MICROPY_VM_HOOK_LOOP` to `RUN_BACKGROUND_TASKS`, and remove the `#ifdef` guards.

My choice of where to put the semicolon is deliberate, so that we can say
```
        RUN_BACKGROUND_TASKS;
```
not have a redundant semicolon, and not confuse C code formatting.

I believe this will be clearer and more flexible in the long run.

tannewt, 2019-07-29:
Works for me!

jepler, 2019-08-11:
In addition, it's technically incorrect to use MICROPY_VM_HOOK_LOOP without the _INIT and _RETURN macros also being present, I think.

jepler, 2019-08-11:
@dhalbert what about MICROPY_EVENT_POLL_HOOK, used in shared-module/bitbangio/SPI.c?

PR #2055 (2019-08-19): Implement and use RUN_BACKGROUND_TASKS in place of MICROPY_VM_HOOK_LOOP
I only touched the areas that I believe are specific to circuitpython, so as to reduce the chances of merge conflicts with upstream, so there are uses of the deprecated (to us) macro remaining.

Closes: #2013
closed 2019-08-19 by dhalbert