adafruit/circuitpython #2021: Docs: Module Support Matrix Updates

sommersoft, 2019-07-31:
Now that the first iteration of the new module support matrix is done, it needs more work. I'm going to put a few issues into this single one; if anyone would rather have them broken up feel free (or let me know).

- [ ] **Text does not wrap in the matrix table:**
  There are some CSS settings that are in `/docs/static/customstyle.css` that should force wrapping inside the table. They are somehow either getting overwritten, or not included.

------------------------

- [ ] **Some modules are not captured correctly by `/docs/shared_bindings_matrix.py`:**
  These _can_ be handled as special cases, or there may be some way to correlate them by just deepening the map. But, I definitely don't subscribe to the notion that the core should bend to the docs.

  - `wiznet`: this module has not been migrated to the "newer" definition system, and relies on 
  `#ifdef`s. As such, it will not show up in the support matrix as is.

  - `fontio`/`terminalio`: these two are included based on `CIRCUITPY_DISPLAYIO`, and do not have their own entries in `circuitpy_mpconfig.mk`. The documentation does not currently provide any indication that `displayio` is required.

   - `socket`: this is included based on `CIRCUITPY_NETWORK`, and does not have it's own entry. Documentation also does not state that `network` is required.

  - `audiocore`: this is included based on `CIRCUITPY_AUDIOIO`, but has its own entry in `cirpy_mpconfig.mk`. As such, it will show up available on all boards because there is nothing to negate it.

--------------------------

- [ ] **Have a separate page for each board:**
  - Initial thought is to add a `/docs/boards` directory where the pages will live.
  - Would love to have them completely built dynamically. May be possible with Jinja and Sphinx templating (I'm not smart enough on templating yet).
  - @tannewt mentioned having each page list "what is present, and what is not".

Neradoc, 2021-03-29:
I'd add to the list of modules that are not captured: `adafruit_bus_device`, recently added to the core, also missing from the matrix, since it's based on `CIRCUITPY_BUSDEVICE`.

Also, the aliased boards are missing from the Support Matrix. These are boards where the build is identical to another, and therefore is not generated on its own, but only listed separately on circuitpython.org and copied to S3. As seen in `build_board_info.py`:
https://github.com/adafruit/circuitpython/blob/main/tools/build_board_info.py#L76-L85

As a tangent, I would suggest migrating from lists in python scripts to a separate configuration file (or files) for the list of aliases, `language_allow_list`, `SUPPORTED_PORTS` and such.

skerr92, 2022-10-16:
Given this issue has had a few PRs merged on it (see above), is it safe to have this one closed?

dhalbert, 2022-10-16:
> Given this issue has had a few PRs merged on it (see above), is it safe to have this one closed?

Could you check on the missing modules mentioned in the second task? If they are present then we could close, otherwise need to add them too.

skerr92, 2022-10-16:
> > Given this issue has had a few PRs merged on it (see above), is it safe to have this one closed?
> 
> Could you check on the missing modules mentioned in the second task? If they are present then we could close, otherwise need to add them too.

It appears only fontio has been captured in the `shared_bindings_matrix.py` https://github.com/adafruit/circuitpython/blob/main/docs/shared_bindings_matrix.py#L67 

The wiznet and socket have not been captured.

dhalbert, 2023-10-28:
I think most of this has been handled in one way or another. Closing. More specific issues can be opened if wanted.

closed 2023-10-28 by dhalbert