adafruit/circuitpython #2023: Hashlib on SAMD

brentru, 2019-07-31:
Hashlib is not supported on SAMD ports of CircuitPython. 

The SHA256 is useful for a RSA implementation I'm working on. SHA1 is useful for [TOTP](https://learn.adafruit.com/circuitpython-totp-otp-2fa-authy-authenticator-friend/software), and MD5 may come in handy for performing checksums for files on CIRCUITPY.

tannewt, 2019-09-03:
Note the SAMD51 has a Public Key Cryptography Controller (PUKCC) that is meant to help with these algorithms.

ladyada, 2019-09-04:
yes! and other chips don't, having generic C is fine and for large chips only

applecuckoo, 2023-05-17:
Related: #4504 

The Integrity Check Monitor (ICM) built into the SAMD51 doesn't support MD5 and neither do the chips I mention in that latest post.