adafruit/circuitpython #2024: Add .hex property to bytes objects

brentru, 2019-07-31:
The CPython bytes.hex() property does not exist for CircuitPython bytes objects:

```
>>> b'\xf0\xf1\xf2'.hex()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'bytes' object has no attribute 'hex'
```

https://docs.python.org/3/library/stdtypes.html#bytes.hex

dhalbert, 2019-07-31:
As a workaround, you can use `binascii.hexlify()`, I think.

brentru, 2019-07-31:
@dhalbert  `binascii` does not exist within samd ports (https://github.com/adafruit/circuitpython/issues/2022). Hexlify also returns a bytes object, .hex returns a string object.

tannewt, 2019-08-02:
@brentru I corrected the terminology here. `.hex()` is a method because it is a function on an object. Properties are used like variables without parens.

theacodes, 2019-10-24:
Here's CPython's implementation: https://github.com/python/cpython/blob/master/Python/pystrhex.c

dhalbert, 2023-10-28:
This now exists, not sure when it showed up:
```py
Adafruit CircuitPython 9.0.0-alpha.2 on 2023-10-27; Adafruit Metro M4 Express with samd51j19
>>> b'abc'.hex()
'616263'
```

closed 2023-10-28 by dhalbert