adafruit/circuitpython #2025: displayio HardFault after restarting an SSD1306-using program

dhalbert, 2019-07-31:
I ran this program, ^C'd it, and then did ^D to do a soft reload. This doesn't always happen, but it did happen several times, so I tried to provoke. Probably not related to BLE. Notice it's crashing inside the import of `adafruit_displayio_ssd1306.mpy`.
```python3
"""CTS demo
"""

import board
import displayio
import terminalio
import time

import adafruit_displayio_ssd1306
from adafruit_display_text import label
from adafruit_ble.current_time_client import CurrentTimeClient

displayio.release_displays()

i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_address=0x3c)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)

FONT = terminalio.FONT
WHITE = 0xffffff
CONNECTING = "Connecting..."
text = CONNECTING

text_area = label.Label(FONT, max_glyphs=50, text=text, color=WHITE)

text_area.x = 0
text_area.y = 10

display.show(text_area)

cts_client = CurrentTimeClient()

while True:
    text_area.text = CONNECTING
    cts_client.start_advertising()
    while not cts_client.connected:
        pass
    cts_client.discover()
    cts_client.pair()

    while cts_client.connected:
        year, month, day, hour, minute, second, _, _, _ = cts_client.current_time
        text_area.text = '{}-{}-{} {:02}:{:02}:{:02}'.format(year, month, day, hour, minute, second)
        time.sleep(5)
```
Backtrace:
```
#0  HardFault_Handler () at supervisor/port.c:138
#1  <signal handler called>
#2  0x0007e400 in displayio_group_get_refresh_areas (self=0x20008490, tail=0x0)
    at ../../shared-module/displayio/Group.c:314
#3  0x0007ceae in displayio_display_get_refresh_areas (self=0x20006b70 <displays+64>)
    at ../../shared-module/displayio/Display.c:230
#4  0x000808be in displayio_refresh_displays () at ../../shared-module/displayio/__init__.c:150
#5  0x0005e33e in run_background_tasks () at background.c:52
#6  0x0004e96a in mp_execute_bytecode (code_state=0x2003f918, inject_exc=<optimized out>) at ../../py/vm.c:1103
#7  0x0003ecfe in fun_bc_call (self_in=0x20008130, n_args=0, n_kw=0, args=0x0) at ../../py/objfun.c:284
#8  0x0003785c in mp_call_function_n_kw (fun_in=0x20008130, n_args=0, n_kw=0, args=0x0) at ../../py/runtime.c:624
#9  0x00037804 in mp_call_function_0 (fun=0x20008130) at ../../py/runtime.c:598
#10 0x0004cb1a in mp_builtin___build_class__ (n_args=3, args=0x2003fac0) at ../../py/modbuiltins.c:60
#11 0x0003eb8a in fun_builtin_var_call (self_in=0x97998 <mp_builtin___build_class___obj>, n_args=3, n_kw=0, 
    args=0x2003fac0) at ../../py/objfun.c:127
#12 0x0003785c in mp_call_function_n_kw (fun_in=0x97998 <mp_builtin___build_class___obj>, n_args=3, n_kw=0, 
    args=0x2003fac0) at ../../py/runtime.c:624
#13 0x0004de38 in mp_execute_bytecode (code_state=0x2003faa8, inject_exc=<optimized out>) at ../../py/vm.c:932
#14 0x0003ecfe in fun_bc_call (self_in=0x200080e0, n_args=0, n_kw=0, args=0x0) at ../../py/objfun.c:284
#15 0x0003785c in mp_call_function_n_kw (fun_in=0x200080e0, n_args=0, n_kw=0, args=0x0) at ../../py/runtime.c:624
#16 0x00037804 in mp_call_function_0 (fun=0x200080e0) at ../../py/runtime.c:598
#17 0x0004bd62 in do_execute_raw_code (module_obj=0x200396e0, raw_code=0x20008420, 
    filename=0x2003fbd4 "adafruit_displayio_ssd1306.mpy") at ../../py/builtinimport.c:177
#18 0x0004bde4 in do_load (module_obj=0x200396e0, file=0x2003fcd4) at ../../py/builtinimport.c:236
#19 0x0004c35a in mp_builtin___import__ (n_args=5, args=0x2003fd74) at ../../py/builtinimport.c:470
#20 0x00038a32 in mp_import_name (name=865, fromlist=0x96d68 <mp_const_none_obj>, level=0x1)
    at ../../py/runtime.c:1392
#21 0x0004e8ae in mp_execute_bytecode (code_state=0x2003fe00, inject_exc=<optimized out>) at ../../py/vm.c:1217
#22 0x0003ecfe in fun_bc_call (self_in=0x200396f0, n_args=0, n_kw=0, args=0x0) at ../../py/objfun.c:284
#23 0x0003785c in mp_call_function_n_kw (fun_in=0x200396f0, n_args=0, n_kw=0, args=0x0) at ../../py/runtime.c:624
#24 0x00037804 in mp_call_function_0 (fun=0x200396f0) at ../../py/runtime.c:598
#25 0x0006243a in parse_compile_execute (source=0x8c7fc, input_kind=MP_PARSE_FILE_INPUT, exec_flags=32, 
    result=0x2003ffb4) at ../../lib/utils/pyexec.c:114
#26 0x00062832 in pyexec_file (filename=0x8c7fc "code.py", result=0x2003ffb4) at ../../lib/utils/pyexec.c:529
#27 0x00053c2c in maybe_run_list (filenames=0x20004000 <supported_filenames>, exec_result=0x2003ffb4)
    at ../../main.c:178
#28 0x00053d34 in run_code_py (safe_mode=NO_SAFE_MODE) at ../../main.c:236
#29 0x000541ba in main () at ../../main.c:445
```
Point of crash:
```
(gdb) fr 2
#2  0x0007e400 in displayio_group_get_refresh_areas (self=0x20008490, tail=0x0)
    at ../../shared-module/displayio/Group.c:314
314	        if (MP_OBJ_IS_TYPE(layer, &displayio_tilegrid_type)) {
(gdb) p layer
$1 = (mp_obj_t) 0x11006c
(gdb) x layer
0x11006c:	0x00000000
(gdb) p *self
$2 = {base = {type = 0x959a4 <closure_type>}, children = 0x20008480, x = 1, y = 0, scale = 33840, size = 8192, 
  max_size = 326, item_removed = false, in_group = false, absolute_transform = {x = 6926, y = 0, dx = -10 '\366', 
    dy = 0 '\000', scale = 0 '\000', width = 33936, height = 8192, mirror_x = false, mirror_y = false, 
    transpose_xy = false}, dirty_area = {x1 = 0, y1 = 0, x2 = 406, y2 = 0, next = 0x1b46}}
(gdb) p self->children
$3 = (displayio_group_child_t *) 0x20008480
(gdb) p *self->children
$4 = {native = 0x965a0 <mp_type_fun_bc>, original = 0x200396d0}
(gdb) p *(self->children)
$5 = {native = 0x965a0 <mp_type_fun_bc>, original = 0x200396d0}
(gdb) 
```

tannewt, 2019-08-02:
@dhalbert this is on master right? Not 4.x. I think I've fixed it in my current work.

dhalbert, 2019-08-02:
right, master. I thought maybe it was due to I2C handling, but I don't remember if that's in 4.x.

tannewt, 2019-08-02:
I think it's related to the changes I made with the oled PR around not sharing group objects between displays. I believe I already found and fixed it with my epaper work.

dhalbert, 2019-08-02:
I'll test it after the epaper PR.

tannewt, 2019-08-23:
This is fixed by #2063

closed 2019-08-23 by tannewt