adafruit/circuitpython #2029: Pygamer Joystick pins

josedlrios, 2019-08-01:
I'm trying the libraries and examples from the bundle and realized the joystick pins are swapped, whenever I move the joystick UP or DOWN the values I receive are from the JOYSTICK_X, same for the y axis (LEFT or RIGHT) is this normal behavior? any suggestion?. 

This is my code:
`import time
import board
import analogio

def read_x(samples=3):
    reading = 0
    for sample in range(0,3):
        reading += x.value
    reading/=3
    return reading

def read_y(samples=3):
    reading = 0
    for sample in range(0,3):
        reading += y.value
    reading/=3
    return reading

x = analogio.AnalogIn(board.JOYSTICK_X)
y = analogio.AnalogIn(board.JOYSTICK_Y)

while True:
    jx=read_x()
    jy = read_y()
    
    print(("x: {}, y: {}").format(jx,jy))
    time.sleep(0.01)`

deshipu, 2019-08-01:
Which version of CircuitPython are you using?

josedlrios, 2019-08-01:
@deshipu 4.0.2

deshipu, 2019-08-01:
Can you try with 4.1.0-rc.1? I know the pins were swapped in the early versions, not sure in which version it got fixed.

josedlrios, 2019-08-01:
@deshipu It works incredible well also the example for cursor_control works smoothly any ideas on why is that? is it better to stick with the unstable version? Many thanks for the help!

deshipu, 2019-08-01:
The pins were swapped by error, and it got fixed in a later version. It should get released as stable soon.

closed 2019-08-01 by josedlrios