adafruit/circuitpython #2033: displayio: pre-allocated buffers and read-only bitmaps

deshipu, 2019-08-03:
I'm thinking about submitting a similar pull request as #2032, but for bitmaps. One thing I would like to add there, though, is the possibility of using a bytes object as the buffer (possibly one that is frozen with your library in flash) — but then the bitmap would need to be read-only, as we can't modify the contents of such a buffer. I think the easiest way to do that would just be to have an internal flag in the ``Bitmap`` object that is set when such a buffer is passed to it, and makes it raise exceptions when a modification is attempted.

Other possibilities would be copy-on-write, or even keeping a list of changes on top of the read-only buffer, but that gets complex really fast and doesn't really add much in practice.

What do you think?

(Also, should we move Bitmap and OnDiskBitmap to ``displaycore`` just like with ``audiocore``?)

tannewt, 2019-08-06:
Bitmap already has read_only: https://github.com/adafruit/circuitpython/blob/master/shared-module/displayio/Bitmap.h#L45 It's used for the blinka in the top left already so it should be easy to extend.

`displaycore` is an interesting idea. It might be worth doing to separate out epaper from mipi support if it ends up too big. Let's revisit that once the epaper PR is closer.

deshipu, 2021-04-02:
So all we need here is to expose it in a constructor.