adafruit/circuitpython #2035: Question - How to remove default shared-modules from build

joshmcintyre, 2019-08-03:
Hi all,

I'm trying to add a custom module that generates Bitcoin addresses to a CircuitPython build. I am building off of the `4.1.x` branch.

However, I'm overflowing the available FLASH space by about 44KB
```
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: build-circuitplayground_express/firmware.elf section `.text' will not fit in region `FLASH'
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 44412 bytes
collect2: error: ld returned 1 exit status
Makefile:326: recipe for target 'build-circuitplayground_express/firmware.elf' failed
make: *** [build-circuitplayground_express/firmware.elf] Error 1
```

I'm trying to trim down my code as much as possible, but even removing some unused functions didn't change the amount I'm going over.

**I'd like to know how to remove some of the builtins I won't use from the build.** I'm having trouble finding anywhere in the atmel-samd makefile or config files that would allow me to remove modules I don't want.

ladyada, 2019-08-03:
youch thats large :) can you use M4 or nRF52 builds only? M0 is super tight fit :/

dhalbert, 2019-08-03:
Look in `mpconfigboard.mk` for various boards and you'll see things like `CIRCUITPY_BLAH = 0` or `=1`. That will turn that particular module on and off. A list of these settings can be gleaned from `circuitpython/py/circuitpy_mpconfig.mk`. But yeah, that's pretty large.

You could also remove some frozen modules if you don't use them, in the same file.

joshmcintyre, 2019-08-03:
Thanks for the responses all!!

@dhalbert thank you, I'll look there.

@ladyada I've been interested in trying the Grand Central so I'd like to get an M4 build working. But when I tried to compile for that, I got an error I can't quite figure out:

``
`/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: build-grandcentral_m4_express/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc_nano.a(lib_a-explicit_bzero.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc_nano.a(lib_a-explicit_bzero.o)
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: error: build-grandcentral_m4_express/firmware.elf uses VFP register arguments, /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc_nano.a(lib_a-bzero.o) does not
/usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/bin/ld: failed to merge target specific data of file /usr/lib/gcc/arm-none-eabi/6.3.1/../../../arm-none-eabi/lib/libc_nano.a(lib_a-bzero.o)
collect2: error: ld returned 1 exit status
Makefile:326: recipe for target 'build-grandcentral_m4_express/firmware.elf' failed
make: *** [build-grandcentral_m4_express/firmware.elf] Error 1
```
Any ideas there?

An aside, thank you both for your work! I'm new to microprocessors (introduced by the PyCon goodie bag this year) and I'm having way too much fun.

-Josh

dhalbert, 2019-08-03:
We are using arm-none-eabi 7.3.1, which you can get from https://launchpad.net/~team-gcc-arm-embedded/+archive/ubuntu/ppa, or directly from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads: use the `7-2018-q2-update` release. Could you try updating your compiler?

joshmcintyre, 2019-08-04:
Hi Dan and Limor, I updated my compiler to the one you've recommended.

After some minor adjustments to code on my end, I was able to compile for the `Grand Central (M4)` board. Looking forward to testing the build when I get that product ordered.

I'll also play around and see if I can get my build it to fit on an `M0` board like the `Circuit Playground` with those tips on removing default modules.

Thanks for all of your help, that resolves the questions I have!

closed 2019-08-04 by joshmcintyre

ladyada, 2019-08-04:
yay!