adafruit/circuitpython #2036: Improve Buffer use in audio APIs

jepler, 2019-08-04:
Pull request #2032 suggests to add a pre-allocated buffer to WaveFile.  However, maybe there's a different route that would be better.  Let's see if we can flesh it out.

It appears that in general, while actually DMA'ing the audio data to the DAC, PWM, or I2C destination, there's an unavoidable need for a chunk of audio data in an arrangement that suits the specific hardware implementation.  For example, on nrf's PWMAudioOut, it is a number of 32-bit blocks, which may contain 1 or 2 channels each of data; the values are rescaled from the original 8- or 16-bit values according to a sample-rate dependent maximum value.  It just so happens that this is always going to be at least as much data as the WaveFile will use, but often more by a factor of 2x or 4x.

What if, by careful processing of the audio data, WaveFile could be given its buffer *by the AudioOut instance using it*, so that it didn't need *any* excess allocations?  This might require allocating a buffer to the max of the original-format sample size and the hardware sample format, and processing carefully so that sample data is overwritten only after it doesn't need to be used anymore.

Would this work with actual dac and i2s implementations?  I did not study the samd implementations in detail, but it looks like their stereo audio (2 distinct DMA channels, which run independently?) could pose a potential problem.

deshipu, 2019-08-04:
How about doing the same thing as in #2032 but for your ``AudioOut`` implementation, if it needs a buffer, so it can be shared?

jepler, 2019-08-04:
What if the sample object can give the AudioOut permission to destroy (vs conserve) the buffer that it gives

deshipu, 2019-08-04:
I can't think of a use case where you would want it conserved?

jepler, 2019-08-04:
RawSample needs it conserved.

tannewt, 2019-08-06:
One tricky thing with a looped sample is knowing whether the sample has already been transformed or not.

kevinjwalters, 2020-07-02:
I think the current nrf implementation does some degree of deallocation which may be useful sometimes but not all of the time. I've created a CircuitPython workaround which appears to have solved my recent spate of 2048 `MemoryError` exceptions: [Adafruit Forums: Audio on nRF52840 with low or fragmented memory](https://forums.adafruit.com/viewtopic.php?f=60&t=166941)