adafruit/circuitpython #2037: gemma_m0 dotstar behaving oddly on boot/REPL entry

jerryneedell, 2019-08-04:
With the current master 5.0 and with 4.1.0 my gemma_m0 seems to  work OK, but the Dotstar LED is behaving oddly on boot and REPL entry.
On boot or reset, it flashes quickly - red, blue, white 
on REPL entry it can be different colors - blue, light blue, light red (pink)  - it may just be whatever the flashing color stopped at.

I have not tried older version of CP.

jerryneedell, 2019-08-18:
I have gemma_m0 that I can now connect to my J-Link, but as in #2064 I am not seeing any obvious reason for the strange behavior. The values sent to the LED appear to be as expected, but the LED is not responding as expected.
Let me know if you have some particular section of code you'd like to see details from.

dhalbert, 2019-08-18:
I can't reproduce this with a couple of Gemma M0's with latest master. Looks the same on MacOS or Linux. I tried an older Gemma M0 and a PyCon 2018 one, which has a dimmer DotStar (the mfr changed them), but they are still just pulsing green

jerryneedell, 2019-08-19:
My apologies -- apparently this sia HW issue with a single Gemma_M0 -- I just tried a another one an it works normally. Closing...

closed 2019-08-19 by jerryneedell

dhalbert, 2019-08-19:
@jerryneedell Did you have another Gemma to test, or are you just going by my tests? It's possible there's something weird with the host USB timing or some such.

jerryneedell, 2019-08-19:
I tried it on another Gemma_m0 and behaves normally.

jerryneedell, 2019-08-19:
This behavior is very different from the issue on the PyRuler as well. The faulty gemma m0 flashes wildly for some time then pulses for awhile then flashes again....  If I can reproduce it, I'll re-open this.

dhalbert, 2019-08-19:
It's certainly weird. I'd expect more trouble with NeoPixels than DotStars.

jerryneedell, 2019-08-19:
hmmm -- looking closely at th malfunctioning Gemma_m0 I can see that the green LED in the DotStar is actually pulsing as expected, but the blue is on steady and the red intermittently starts to flash on/f very quickly.  It does look like a HW issue with this board/dotstar. I tried pressing on it, but no change. If I get brave, I may try replacing it. For now, I'll move onto other problems ....

jerryneedell, 2019-09-02:
fixed by #2102