adafruit/circuitpython #2038: Saving code to main.py does not reload and run

hybotics, 2019-08-05:
Saving an updated main.py is supposed to force a reload and run the new code with any errors reported in the REPL.

What happens is when I save a new main.py, I get an OS error 5 in the REPL. I have to press Ctrl/D in the REPL many (usually over 30) times to get anything to happen. Until it takes off, I get the OS error 5 each time I press Ctrl/D. The OS error 5 is not valid here.

dhalbert, 2019-08-05:
Is this true of any `main.py` or just this particular one? If the latter, please post the code here. If not, try reformatting the CIRCUITPY filesystem:
```
import storage
storage.erase_filesystem()
```

hybotics, 2019-08-05:
It is true *any* time I save a new main.py. I am running Raspbian Stretch with the Geany editor. I will try reformatting the CIRCUITPY drive and see if that helps.

dhalbert, 2019-08-05:
This may be due to the geany editor not flushing all the file writes immediately. I haven't tested it or reviewed its source code to see if it does. See https://learn.adafruit.com/welcome-to-circuitpython/creating-and-editing-code#1-use-an-editor-that-writes-out-the-file-completely-when-you-save-it-7-13

hybotics, 2019-08-05:
I saved all the files from my board and reformatted the storage. That fixed the problem! Thanks for the suggestion.

closed 2019-08-05 by hybotics

dhalbert, 2019-08-05:
Great! Do think about whether geany is the problem: if it does only a partial write and you do a hardware reset, the filesystem will get corrupted. It looks like geany uses Gnome file operations, and I don't see that those do something like an `fsync`. You could do a `sync` from the shell after a write to make sure the file is completely written.