adafruit/circuitpython #2041: Polish up native api docs so that stubs are complete

tannewt, 2019-08-06:
`make stubs` generates .pyi files that can be used in editors like VS code to provide helpful hints. The rST docs need to be fixed up so that the stubs include things like valid types.

tannewt, 2019-08-06:
PR that added stubs: https://github.com/adafruit/circuitpython/pull/1915

Example PR that fixed up some comments: https://github.com/adafruit/circuitpython/pull/1920

sw23, 2020-10-28:
@tannewt Can I pick up this work item for #hacktoberfest?  I looked at the code and have a good handle on what needs to be done.  Thanks!

tannewt, 2020-10-28:
@sw23 Totally! Feel free to submit PRs to improve the type stubs.

dhalbert, 2021-06-15:
@dherrada Is this kind of all done?

dherrada, 2021-06-16:
@dhalbert I think so, but @tannewt would know best

tannewt, 2021-06-22:
I think so too. I suspect we'll get issues once more folks use them but we can have separate issues/PRs for those. Thanks @dherrada for going over it all.

closed 2021-06-22 by tannewt