adafruit/circuitpython #2045: Pyportal hard faults

jerryneedell, 2019-08-07:
I wish I had more specific information , but I am getting repeated Hard Fault crashes to Safe mode on a PyPortal with the current Master.
```Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

Press any key to enter the REPL. Use CTRL-D to reload.

```
After a hard reset it seems to work fine with a variety of programs but then after a soft reboot (or after a few) it will crash rerunning a program that has run previously.

Just wanted to get this posted as a heads up. I have not seen it on other boards...yet.

an example that runs after hard reset, but crashes is run after a soft reboot is the paint demo
```
"""
Paint for PyPortal, PyBadge, PyGamer, and the like.

Adafruit invests time and resources providing this open source code.
Please support Adafruit and open source hardware by purchasing
products from Adafruit!

Written by Dave Astels for Adafruit Industries
Copyright (c) 2019 Adafruit Industries
Licensed under the MIT license.

All text above must be included in any redistribution.
"""

#pylint:disable=invalid-name, no-self-use

import gc
import time
import board
import displayio
import adafruit_logging as logging
try:
    import adafruit_touchscreen
except ImportError:
    pass
try:
    from adafruit_cursorcontrol.cursorcontrol import Cursor
    from adafruit_cursorcontrol.cursorcontrol_cursormanager import DebouncedCursorManager
except ImportError:
    pass

class Color(object):
    """Standard colors"""
    WHITE = 0xFFFFFF
    BLACK = 0x000000
    RED = 0xFF0000
    ORANGE = 0xFFA500
    YELLOW = 0xFFFF00
    GREEN = 0x00FF00
    BLUE = 0x0000FF
    PURPLE = 0x800080
    PINK = 0xFFC0CB

    colors = (BLACK, RED, ORANGE, YELLOW, GREEN, BLUE, PURPLE, WHITE)

    def __init__(self):
        pass

################################################################################

class TouchscreenPoller(object):
    """Get 'pressed' and location updates from a touch screen device."""
[... 337 lines cut ...]
            y_changed = self._location[1] != self._last_location[1]
            return x_changed or y_changed
        else:
            return False

    def _update(self):
        self._last_a_pressed, self._last_location = self._a_pressed, self._location
        self._a_pressed, self._location = self._poller.poll()

    def run(self):
        """Run the painting program."""
        while True:
            self._update()
            if self._was_a_just_pressed:
                self._handle_a_press(self._location)
            elif self._was_a_just_released:
                self._handle_a_release(self._location)
            if self._did_move and self._a_pressed:
                self._handle_motion(self._last_location, self._location)
            time.sleep(0.1)

painter = Paint()
painter.run()

```

jerryneedell, 2019-08-07:
just to be sure, I recloned the CP repo and rebuilt sot remove the "dirty" buult -- it was not getting the correct tag for tinyusb. rebuilt and still have same crashes 
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-alpha.0-153-g47a0b7cba on 2019-08-07; Adafruit PyPortal with samd51j20
>>> 

```
trying to find a simpler test case....

jerryneedell, 2019-08-08:
here is another way to trigger the Hard fault:
Hard reset:
run slide show:
```
import os
import board
import busio
import digitalio
import storage
import adafruit_sdcard
from adafruit_slideshow import PlayBackOrder, SlideShow

""" If you want to use an SD card to store the images: """
spi = busio.SPI(board.SCK, MOSI=board.MOSI, MISO=board.MISO)
cs = digitalio.DigitalInOut(board.SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

IMAGE_DIRECTORY = "/sd/images"
#IMAGE_DIRECTORY = "/images"

def print_directory(path, tabs=0):
    for file in os.listdir(path):
        stats = os.stat(path + "/" + file)
        filesize = stats[6]
        isdir = stats[0] & 0x4000

        if filesize < 1000:
            sizestr = str(filesize) + " by"
        elif filesize < 1000000:
            sizestr = "%0.1f KB" % (filesize / 1000)
        else:
            sizestr = "%0.1f MB" % (filesize / 1000000)

        prettyprintname = ""
        for _ in range(tabs):
            prettyprintname += "   "
        prettyprintname += file
        if isdir:
            prettyprintname += "/"
        print('{0:<20} Size: {1:>6}'.format(prettyprintname, sizestr))

        # recursively print directory contents
        if isdir:
            print_directory(path + "/" + file, tabs + 1)

print_directory(IMAGE_DIRECTORY)

# Create the slideshow object that plays through once alphabetically.
slideshow = SlideShow(board.DISPLAY, None,
            folder=IMAGE_DIRECTORY, loop=True,
            order=PlayBackOrder.RANDOM, dwell=2.5)

while slideshow.update():
    pass

```
then soft reboot (control-d)  and run this "simpletest"
``
# NOTE: Make sure you've created your secrets.py file before running this example
# https://learn.adafruit.com/adafruit-pyportal/internet-connect#whats-a-secrets-file-17-2
import board
from adafruit_pyportal import PyPortal

# Set a data source URL
TEXT_URL = "http://wifitest.adafruit.com/testwifi/index.html"

# Create the PyPortal object
pyportal = PyPortal(url=TEXT_URL, status_neopixel=board.NEOPIXEL)

# Set display to show REPL
board.DISPLAY.show(None)

# Go get that data
print("Fetching text from", TEXT_URL)
data = pyportal.fetch()

# Print out what we got
print('-'*40)
print(data)
print('-'*40)

```

Either of these can be run repeatedly after sof reboots, , but the "simpletest" crashes if run after the slideshow....

jerryneedell, 2019-08-08:
another example is the new Wake Up light demo https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/PyPortal_Wakeup_Light

It runs fine after a hard reset, but crashes to Safe Mode  if restarted after after a soft reboot.

jerryneedell, 2019-08-08:
hmm -- I get the same behavior with CP 4.1.0

jerryneedell, 2019-08-08:
correction to now deleted post... I cannot reproduce this issue with CP 4.1.0

jerryneedell, 2019-08-08:
ok -- much simpler example
```
import board
import displayio
import terminalio
from adafruit_display_text import label
 
display = board.DISPLAY
 
# Set text, font, and color
text = "HELLO WORLD"
font = terminalio.FONT
color = 0x0000FF
 
# Create the tet label
text_area = label.Label(font, text="HELLO WORLD", color=0x00FF00)
 
# Set the location
text_area.x = 100
text_area.y = 80
 
# Show it
display.show(text_area)
```

runs ok after hard reset

crashes to hard fault safe mode after a soft reset (control - d)

```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

You are running in safe mode which means something unanticipated happened.
Looks like our core CircuitPython code crashed hard. Whoops!
Please file an issue at https://github.com/adafruit/circuitpython/issues
 with the contents of your CIRCUITPY drive and this message:
Crash into the HardFault_Handler.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-alpha.0-153-g47a0b7cba on 2019-08-07; Adafruit PyPortal with samd51j20
>>> 

```

jerryneedell, 2019-08-08:
I wondered if this was related to #1913, but since it is only  for CP5.0 and so far on PyPortal, I think it may be a different issue.  If anyone has a PyPortal with CP 5.0 installed, it would be nice to have a confirmation using the short test in the previous message.

DavePutz, 2019-08-09:
Confirmed that the test above fails using CP 5.0. I'll see if I can get my JLink hooked up to the PyPortal and get some stack(s).

DavePutz, 2019-08-13:
With the current 5.0 master, I now see slightly different symptoms. The program runs after a hard reset, but also runs after a soft reset(Ctrl-D). However, if I run "exec(open('main.py').read())" from the REPL prompt, it dies with the following stack:

	 HardFault_Handler Line: 297	
 	 <signal handler called> Line: 297
         displayio_group_get_refresh_areas (displayio_group_t * self, displayio_area_t * tail) Line: 313	
 	 displayio_display_get_refresh_areas (displayio_display_obj_t * self) Line: 230	
 	 displayio_display_get_refresh_areas (displayio_display_obj_t * self) Line: 225	
 	 displayio_refresh_displays Line: 150	
 	 displayio_refresh_displays Line: 115	
 	 run_background_tasks Line: 63	
 	 run_background_tasks Line: 51	
 	 mp_execute_bytecode (mp_code_state_t * code_state, mp_code_state_t * code_state@entry, volatile mp_obj_t inject_exc, volatile mp_obj_t inject_exc@entry) Line: 1103	
 	 fun_bc_call (mp_obj_t self_in, size_t n_args, size_t n_kw, const mp_obj_t * args) Line: 284	
 	 mp_builtin___build_class__ (size_t n_args, const mp_obj_t * args) Line: 60	
 	 fun_builtin_var_call (mp_obj_t self_in, size_t n_args, size_t n_kw, const mp_obj_t * args) Line: 127	
 	 mp_call_function_n_kw (const mp_obj_t * args, size_t n_args, mp_obj_t fun_in) Line: 613	
 	 mp_execute_bytecode (mp_code_state_t * code_state, mp_code_state_t * code_state@entry, volatile mp_obj_t inject_exc, volatile mp_obj_t inject_exc@entry) Line: 932	
 	 fun_bc_call (mp_obj_t self_in, size_t n_args, size_t n_kw, const mp_obj_t * args) Line: 284	
 	 do_execute_raw_code (const char * filename, mp_raw_code_t * raw_code, mp_obj_t module_obj) Line: 177	
 	 do_load (mp_obj_t module_obj, mp_obj_t module_obj@entry, vstr_t * file, vstr_t * file@entry) Line: 236	
 	 mp_builtin___import__ (size_t n_args, size_t n_args@entry, const mp_obj_t * args, const mp_obj_t * args@entry) Line: 470	
 	 mp_import_from (mp_obj_t module, qstr name) Line: 1439	
 	 mp_execute_bytecode (mp_code_state_t * code_state, mp_code_state_t * code_state@entry, volatile mp_obj_t inject_exc, volatile mp_obj_t inject_exc@entry) Line: 1224	
 	 fun_bc_call (mp_obj_t self_in, size_t n_args, size_t n_kw, const mp_obj_t * args) Line: 284	
 	 mp_parse_compile_execute (mp_lexer_t * lex, mp_parse_input_kind_t parse_input_kind, mp_parse_input_kind_t parse_input_kind@entry, mp_obj_dict_t * globals, mp_obj_dict_t * globals@entry, mp_obj_dict_t * locals, mp_obj_dict_t * locals@entry) Line: 1488	
 	 eval_exec_helper (size_t n_args, const mp_obj_t * args, mp_parse_input_kind_t parse_input_kind) Line: 148	
 	 fun_builtin_var_call (mp_obj_t self_in, size_t n_args, size_t n_kw, const mp_obj_t * args) Line: 127	
 	 mp_call_function_n_kw (const mp_obj_t * args, size_t n_args, mp_obj_t fun_in) Line: 613	
 	 mp_execute_bytecode (mp_code_state_t * code_state, mp_code_state_t * code_state@entry, volatile mp_obj_t inject_exc, volatile mp_obj_t inject_exc@entry) Line: 932	
 	 fun_bc_call (mp_obj_t self_in, size_t n_args, size_t n_kw, const mp_obj_t * args) Line: 284	
 	 parse_compile_execute (const void * source, const void * source@entry, mp_parse_input_kind_t input_kind, int exec_flags, int exec_flags@entry, pyexec_result_t * result, pyexec_result_t * result@entry) Line: 114	
 	 pyexec_friendly_repl Line: 518	
 	 run_repl Line: 386	
 	 main Line: 438

The contents of self look quite funky - memory has been stepped on somewhere:
-		self	0x20003570	displayio_group_t *
+		  base		mp_obj_base_t
+		  children	0x74726174	displayio_group_child_t *
		    x	   28265	int16_t
		    y	    8807	int16_t
		    scale	8233	uint16_t
		    size	2573	uint16_t
		    max_size	26980	uint16_t
		    item_removed	115	_Bool
		    in_group	112	_Bool
+		  absolute_transform		displayio_buffer_transform_t
+		  dirty_area		displayio_area_t
+		tail	0x0	displayio_area_t *

Not being real familiar with the displayio code I'll have to do some code reading before any of this makes sense. But, a theory would be that we are not clearing out some memory when exec'ing that we do when a hard reset is done.

tannewt, 2019-08-13:
Oh, I think I've fixed this in my epd branch: https://github.com/tannewt/circuitpython/tree/epd

IIRC it's an issue of not resetting the group of a display properly: https://github.com/tannewt/circuitpython/commit/312c2408de359f2fe5324a5f29562d1479d7b415#diff-4ce49108868950347d358317f36c0935L238

In the OLED PR I lumped all of the reset logic into one loop not realizing that the continues would skip the display reset bits when hit.

jerryneedell, 2019-08-13:
@tannewt I tried your EPD branch on my PyPortal, but it always goes to safe mode (HardFauult) on hard reset.   Do I need to make some changes to it  for the PyPortal?

jepler, 2019-08-16:
@tannewt is it possible that this was fixed by the merge of #2063?  I reproduced the issue at 583392a5c694c082f6 but not at 485dd8a67.

jerryneedell, 2019-08-16:
fixed by #2063

closed 2019-08-16 by jerryneedell