adafruit/circuitpython #2056: Traceback from exceptions accumulates on REPL on 4.1.0

kevinjwalters, 2019-08-12:
Observed on `4.1.0-rc.1` and briefly discussed on Discord. `4.1.0` has same issue with accumulation of stack trace stuff in `Traceback` output:

```
Adafruit CircuitPython 4.1.0 on 2019-08-03; Adafruit Metro M4 Express with samd51j19
>>> from adafruit_midi.program_change import ProgramChange
>>> ProgramChange(200) ### intentional illegal value
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_midi/program_change.py", line 56, in __init__
ValueError: Out of range
>>> ProgramChange(200) ### intentional illegal value
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_midi/program_change.py", line 56, in __init__
  File "<stdin>", line 1, in <module>
  File "adafruit_midi/program_change.py", line 56, in __init__
ValueError: Out of range
>>> ProgramChange(200) ### intentional illegal value
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_midi/program_change.py", line 56, in __init__
  File "<stdin>", line 1, in <module>
  File "adafruit_midi/program_change.py", line 56, in __init__
  File "<stdin>", line 1, in <module>
  File "adafruit_midi/program_change.py", line 56, in __init__
ValueError: Out of range
```

kevinjwalters, 2019-08-14:
This seemed fairly harmless in REPL. Is it definitely REPL only? If not, is there any chance of leaking memory for applications with repeatedly/many caught exceptions? I'd guess not as the TraceBack/text would only be prepared when it's not caught?

tannewt, 2019-08-16:
I assume it's REPL because it's the only case where an exception doesn't cause a VM reset. However, I haven't looked in detail yet so I may be wrong.

tannewt, 2020-12-04:
This is due to the MIDI library reusing ValueError: https://github.com/adafruit/Adafruit_CircuitPython_MIDI/blob/master/adafruit_midi/midi_message.py#L108-L109 I'll move this issue to the midi repo.

Instead, it should create a new exception each time.