adafruit/circuitpython #2058: Track DMA channel in use, don't rely on dma_channel_enabled (on atmel-samd)

jepler, 2019-08-13:
While investigating #1908, I observed the following:

---

Next, I started tracking how dma_channel_enabled tracked with calls to dma_{en,dis}able_channel. I found that, contrary to what the code in find_free_audio_dma_channel requires, it appears that channel->CHCTRLA.bit.ENABLE can be set to false asynchronously.

When this happens, it's possible for both AudioOut and PDMIn to simultaneously believe they have been allocated the same DMA channel, and to make conflicting uses of it.

I did not find any other sites in circuitpython that look like they would reset ENABLE, and the datasheet at best fails to state clearly that this bit can be reset automatically at the end of a DMA operation, but one of these two things must be happening.

My recommendation, and I can clean up and PR my local code that is doing this (there are also a lot of inappropriate print statements and so on), is to add a new array of bool that tracks whether CP believes the DMA channels are in use, and use that in find_free_audio_dma_channel. This involves creating new wrapper functions audio_dma_enable_channel and audio_dma_disable_channel that call the SDK functions but also manipulate the new variable: f9e4b99

---

tannewt, 2019-08-13:
👍 from me

dhalbert, 2019-08-29:
Do you think this is related to #2005 as well?

jepler, 2019-08-29:
@dhalbert hmm if so I'm not casting my net far enough. I am only looking at audio dma but in that issue display dma may be implicated too...

commit 30a93463 (2019-09-05): samd: audio_dma: Track channel allocation

Previously, we depended on allocated channels to always be
"dma_channel_enabled".  However, (A) sometimes, many operations
would take place between find_free_audio_dma_channel and
audio_dma_enable_channel, and (B) some debugging I did led me to believe
that "dma_channel_enabled" would become false when the hardware ended
a scheduled DMA transaction, but while a CP object would still think it
owned the DMA channel.

((B) is not documented in the datasheet and I am not 100% convinced that
my debugging session was not simply missing where we were disabli [... cut]
closed 2019-09-05 by tannewt